prusa3d/PrusaSlicer #6503: Multimaterial.tisk - vzor prvni vrstvy pro ruzne extrudery

misisl, 2021-05-09:
### Version
PrusaSlicer-2.3.1+win64-202104161404

### Operating system type + version
Windows 10 (build 19042), 64-bit edition

### 3D printer brand / version + firmware version (if known)
MK3SMMU2

### Behavior
* Pri multimaterial.tisku nelze tisknout prvni vrstvu jinym vzorem pro ruzne materialy.  Pokud to zkusim obejit zmenou vzoru vyplne (koncentricka, 100%) a nastavenim spodnich vrstev na nulu, vzor se na prvni vrstvu nepouzije pokud neni nastaben na nulu i pocet vrchnich vrstev. Pak je zase ignorovan ironing.

* _Expected Results_: Spodni vrstva "okenek" (prusvitny plast = svetle modry extruder) koncentricka, vrchni vrstva monotoni + ironing
* _Actual Results_: vsechny vrstvy koncentricke, ironing ignorovan

[image]

#### Project File (.3MF) where problem occurs
[PC - vypln_mrizky.zip](https://github.com/prusa3d/PrusaSlicer/files/6447320/PC.-.vypln_mrizky.zip)

rtyr, 2021-05-13:
Můžeme issue zavřít?

rtyr, 2021-05-31:
Bez reakce, zavírám.

closed 2021-05-31 by rtyr