prusa3d/PrusaSlicer #6504: Issues when second file is opened.

DerDave11, 2021-05-09:
### Version
2.3.0+

### Operating system type + version
MacBook Pro 2018, MacOS 11.3

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* PrusaSlicer hangs up once I open a second .stl file. 
* 
* _Steps needed to reproduce the problem_
* PrusaSlicer is already open with .stl file loaded. 
* Open another .stl file
* PrusaSlicer will hang up, not able to close it via Close button
* Close via "Quit PrusaSlicer" in top menus bar
* 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-05-19:
Does it happen with all STL files? Could you try it with the latest 2.3.1 release?

DerDave11, 2021-05-20:
Seems like the issue is fixed with 2.3.1. Thank you!

closed 2021-05-20 by DerDave11