prusa3d/PrusaSlicer #6505: Please show current file name somewhere

icehawk1, 2021-05-09:
### Version
Prusa Slicer 2.3.1

### Operating system type + version
Windows 10 

### Behavior
AfterI opened an .3mf file, PrusaSlicer doesnt show the filename anywhere. I have a bunch of projects where I split the parts into multiple print runs and therefore multiple .3mf files. Now, I always have to remember which file I am currently editing. Would be helpful if the slicer could display that somewhere. Lower right corner doesnt seem to be used, yet. :)

rtyr, 2021-05-09:
It should be visible here:
[image]

Or it can be the same issue as described here https://github.com/prusa3d/PrusaSlicer/issues/6473 (it should be already fixed on master).

icehawk1, 2021-05-09:
Ah, I understand. Yes, it is the same issue as #6473

rtyr, 2021-05-09:
> Ah, I understand. Yes, it is the same issue as #6473

I am closing this issue as a duplicate then.

closed 2021-05-09 by rtyr

MucaroBoricua, 2021-09-19:
This issue should not have been closed because #6473 was already closed and the bug was not fixed!  That's what @icehawk1 meant.

rtyr, 2021-09-19:
> This issue should not have been closed because #6473 was already closed and the bug was not fixed! That's what @icehawk1 meant.

- this issue was closed because it was a duplicate of 6473.
- 6473 was closed because the issue was fixed
- the fix is included in 2.4.0-alpha1 as stated here https://github.com/prusa3d/PrusaSlicer/issues/6473#issuecomment-853897625

MucaroBoricua, 2021-09-19:
I didn't know the PrusaSlicer developers would close a problem report **before** releasing a fix and without mentioning the version where it was fixed.  That blindsided me.  Thanks for the clarification, @rtyr.