prusa3d/PrusaSlicer #6510: CTRL-Z doesn't work in PrusaSlicer to Undo

todsandberg, 2021-05-10:
### Version
Version 2.3.1+win64  

### Operating system type + version
Windows 10 Pro   OS Build 19041.928

### 3D printer brand / version + firmware version (if known)
Not Applicable (Prusa MK3s+)

### Behavior
* _Describe the problem_ 
While Using PrusaSlicer I press "CTRL-Z" to undo the last command

* _Expected Results_ 
The last action to be undone

* _Actual Results_ 
The last action is not undone

_Note: selecting UNDO from the menu DOES work "Edit --> Undo"_

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs
This applies on any file that I'm working on.

fe60, 2021-05-11:
Undo works for you from menu.
What about the Undo button in the top menu [1]. Does it work?

[1] https://help.prusa3d.com/en/article/ui-overview_1766 (Undo/Redo is missing on the screenshot)

bubnikv, 2021-05-13:
> What about the Undo button in the top toolbar [1]. Does it work?

Where is your keyboard focus?

Ctrl-Z may be captured by various input fields and dialogs. You have to
have a keyboard focus captured by the 3D scene for Undo to work there.

todsandberg, 2021-05-13:
Thank you. I think I figured it out (or at least a big piece of it).

Try this:

STEPS:
1. On a blank print bed add a shape (box)
2. press CTRL Z.

EXPECTED: the shape to be removed.
ACTUAL: the shape is not removed.

It seems that there are multiple steps recorded in the "Undo History" which
shows multiple steps (2-3) for each item added.
(it adds these additional steps called "Selection-Add from list"
"Selection-Remove All") when you have multiple items added.

I think that may be the real issue, rather than CTRL-Z not working.
Is there something that could be done to make it work as expected?

Thanks.

bubnikv, 2021-05-13:
If you click the Undo button at the top toolbar with your right mouse
button, you will see a pop-up with a list of Undo actions saved.

bubnikv, 2021-09-30:
I believe the issue was explained.
Closing.

closed 2021-09-30 by bubnikv