prusa3d/PrusaSlicer #6513: Useless solid infill on top of and beneath inclined wall

fe60, 2021-05-12:
### Version
 2.3.1+win64

### Behavior
Slicing a model with inclined walls leads to solid infill that seems to be useless (especially when it's sliced with 0% infill).
Is there a special reason for this solid infill on top of and beneath inclined walls?
I expected the simplified dummy object to be printed without the solid infill.

[image][image]

_Is this a new feature request?_
No

stl-file generated from model
[UselessTopInfill.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6465173/UselessTopInfill.stl.zip)

neophyl, 2021-05-12:
First of all when posting issues please include the requested project file (File>Save Project as).  The. 3mf produced contains the model and all your settings which is important to duplicating your issue.

However with your model and the fact you mention 0% infill and I can see 5 perimeters from your picture.  If I set a 'normal' mk3 0.2 speed profile and change infill to 0% and perimeter to 5 I can duplicate your screen cap in appearance.  However if you untick Ensure vertical shell thickness the unwanted infill is no longer produced.  
I suspect this is down to your settings rather than the software.  Do you have Ensure vertical shell thickness enabled ?

Secondly this issue is a duplicate of so many existing issues.  For example https://github.com/prusa3d/PrusaSlicer/issues/1054 or https://github.com/prusa3d/PrusaSlicer/issues/668 .

Please consider searching for existing issues before opening new ones.  The devs have enough to do without trawling through issues that frankly aren't.

fe60, 2021-05-12:
@neophyl was right, it is related to the "Ensure vertical shell tickness" option. Deactivating it makes the solid infill disappear.

I didn't expect settings to be relevant for this behavior (started playing around with the inclination of the walls instead).
Hence I did not attach the .3mf file. I will change this in the future.
I searched for similar issues before creating this one, but obviously I used the wrong keywords.

closed 2021-05-12 by fe60