prusa3d/PrusaSlicer #6514: SLA manual support editing crashes when attempting to remove a support point

davidkerny, 2021-05-13:
Version: 2.3.1+win64
Build: PrusaSlicer-2.3.1+win64-202104161404
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition

When slicing for SL1, 
SLA support points [L]>Manual editing,  then you select a support point, hit BACKSPACE key on your keyboard. Slicer asks if you want to apply changes. 
If you click yes, Slicer crashes.
If you click no, Slicer asks you the same again and then crashes.

Delete key deletes the support point as expected.

lukasmatena, 2021-05-14:
Thanks for the report. I can reproduce the problem in 2.3.1 (Linux), but not in current master. I don't remember the change that would fix it, but it seems to be fine. We'll test on other platforms to be sure.

lukasmatena, 2021-05-14:
@FidelCapo has tested on Windows and macOS, there were no issues (in current master). Closing.

closed 2021-05-14 by lukasmatena