prusa3d/PrusaSlicer #6515: Tiskárna Ender3 s deskou SKR Mini E3 v.2, se v průběhu tisku zastavuje, tisk se zastaví a tiskárna se restartuje a stojí. Problém začal u verze 2.3.0 a u verze 2.3.1 pokračuje. Když g-kód vygeneruji na jiném sliceru k zastavování nedochází. Do verze 2.3,.0 fungoval slicer bez problému, nyní se nedá používat.

gems007, 2021-05-14:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-05-14:
Obávám se, že podle popisu bude problém téměř jistě na straně tiskárny.

Poprosím o nějaký 3MF projekt, se kterým se to stalo.

gems007, 2021-05-14:
bohužel stává se to u všech objektů, tisk se zastaví v různých částech
tisku, tryska zůstává přilepená na místě přerušení tisku a tiskárna se
restartuje a nic se neděje. Nejdřív jsem řešil, že problém je v SD kartě,
výměna nepomohla, potom řídící deska taky pořád stejné až jsem dal dotaz do
fóra na FB a tam jsem se dozvěděl, že to dělá poslední verze sliceru, tak
jsem to ověřil a opravdu, když ten samý soubor rozřežu na sliceru Cura, tak
k zastavování nedochází. Je zajímavé, že toto dělá jen na tiskárně Ender3 s
deskou SKR mini E3 v.2, na tiskárně kde mám starší desku SKR mini E3 v 1.2
to ten samý soubor vytiskne bez problému.

rtyr, 2021-05-14:
Ano, ale stále je to s největší pravděpodobností problém na straně HW/FW, což koneckonců i sám potvrzujete tím, že na starší desce se to neděje.

Prosím o nějaký ten projekt (je nutno vložit přes web interface a předtím zazipovat), abych viděl, jaké používáte nastavení.

rtyr, 2021-05-14:
Nebo stačí g-code.

gems007, 2021-05-14:
tiskárna se pokaždé zastaví na jiném místě tisku. Do verze 2.3.0 jsem tento
slicer používal bez problému, fungoval skvěle. Nejdřív jsem hledal problém
v tiskárně, výměna SD karty, výměna desky a problém stále stejný, když dám
tisknout stejný soubor z jiného sliceru, tak k problémům nedochází.

rtyr, 2021-05-14:
Ano, to už jste zmiňoval, ale to neznamená, že je problém na straně sliceru. Používáte funkci "avoid crossing perimeters"?

gems007, 2021-05-14:
ne tuto funkci nemám zapnutou

rtyr, 2021-05-14:
Až budete moct, tak pošlete nějaká data, jak jsem zmiňoval. Pro porování můžete poslat i váš g-code z jiného sliceru.

Obávám se ale, že problém je třeba řešit jinde, než v PrusaSliceru. Že je g-code v pořádku jste už sám ověřil tiskem toho samého g-codu na starší revizi desky.

rtyr, 2021-06-07:
Bez reakce. Zavírám.

closed 2021-06-07 by rtyr