prusa3d/PrusaSlicer #6516: Segmentation fault (GLib-GObject null pointer)

mdjurfeldt, 2021-05-14:
### Version

version_2.4.0-alpha0-349-ge0dd7edb2

### Operating system type + version

Linux Debian Bullseye

### Behavior

I get the following error message when slicing the attached 3mf file:

(prusa-slicer:29812): GLib-GObject-CRITICAL **: 11:42:10.191: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Segmentation fault

#### Project File (.3MF) where problem occurs
[cam_clip_3_015tol_ASA_mdj.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/6478022/cam_clip_3_015tol_ASA_mdj.3mf.gz)

lukasmatena, 2021-05-14:
Thanks. I reproduced the problem in current master (221af991c2474a8f382042b667269e3346dab25e), but not in 2.3.1. The crash is not related to the GTK assert, it really happens in
```
1   std::vector<Slic3r::ExPolygon>::begin                                                                                                  stl_vector.h    708  0x5e2125  
2   Slic3r::area                                                                                                                           ExPolygon.hpp   368  0x15e130c 
3   Slic3r::GUI::Preview::update_layers_slider                                                                                             GUI_Preview.cpp 681  0x15ddbe1 
4   Slic3r::GUI::Preview::load_print_as_fff                                                                                                GUI_Preview.cpp 922  0x15df806 
5   Slic3r::GUI::Preview::load_print                                                                                                       GUI_Preview.cpp 344  0x15dbf87 
6   Slic3r::GUI::Preview::reload_print                                                                                                     GUI_Preview.cpp 376  0x15dc05e 
7   Slic3r::GUI::Plater::priv::update_fff_scene                                                                                            Plater.cpp      3075 0x10be0e2 
8   Slic3r::GUI::Plater::priv::on_slicing_completed
```

There is an out-of-bounds access on line `double prev_area = area(object->get_layer(i)->lslices);` in `Slic3r::GUI::Preview::update_layers_slider`: i=1 and object->m_layers only contain one element.

bubnikv, 2021-05-14:
Thanks for heads up. Fixed with 2b9e41e6955f2bdf5a1a498bb96f6173a4780ee1

closed 2021-05-14 by bubnikv