prusa3d/PrusaSlicer #6519: Arrange places objects on top of eachother

mdjurfeldt, 2021-05-16:
### Version

version_2.4.0-alpha0-353-g4aab49f59

### Operating system type + version

Debian Bullseye

### Behavior

When I load the attached .3mf file and type "a" (arrange), the objects are placed on top of eachother instead of side by side.

#### Project File (.3MF) where problem occurs

[Clamp Frame+Bolts.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/6488901/Clamp.Frame%2BBolts.3mf.gz)

rtyr, 2021-05-17:
Why do you think it should be placed side by side? What is wrong with this arrangement?

[image]

mdjurfeldt, 2021-05-17:
The advantage of placing the objects side by side, *as in your example* which is not wrong, is that they don't get fused together as they are when using the version of PrusaSlicer which I indicated above. See atached picture.
[image]

rtyr, 2021-05-17:
I see, we are investigating it.

rtyr, 2021-05-18:
Fixed. Closing.

closed 2021-05-18 by rtyr