prusa3d/PrusaSlicer #6523: M600 - big drop of filament after the bresume

Zazou89, 2021-05-16:
### Version
_Version of PrusaSlicer used goes here_ 

PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,098MB
OpenGL installation
GL version:   4.6.14831 Compatibility Profile Context 21.5.1 27.20.21003.1028
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 6800 XT
GLSL version: 4.60

### 3D printer brand / version + firmware version (if known)
Artillery Genius with Molise firmware

### Behavior
I added a manual color chang in PS at layer 3. After the color remplacement, nozzle purges the filament where it should resume. It creates a big drop.

Gcode around M600 : 
```
;WIPE_START
G1 F6240.000
G1 X123.656 Y123.191 E-0.29279 ; wipe and retract
G1 X123.656 Y122.610 E-0.18573 ; wipe and retract
G1 X122.610 Y123.656 E-0.47262 ; wipe and retract
G1 X122.029 Y123.656 E-0.18573 ; wipe and retract
G1 X123.507 Y122.178 E-0.66813 ; wipe and retract
;WIPE_END
G1 E-0.09500 F2100.000 ; retract
G1 Z1.100 F7800.000 ; lift Z
;AFTER_LAYER_CHANGE
;0.7
;COLOR_CHANGE,T0
M600
; printing object xyzCalibration_cube.stl id:0 copy 0
G1 X123.961 Y123.961 ; move to first perimeter point
G1 Z0.700 ; restore layer Z
G1 E1.90000 F2100.000 ; unretract
M204 S800 ; adjust acceleration
;TYPE:Perimeter
;WIDTH:0.45
G1 F2700.000
G1 X106.039 Y123.961 E0.60663 ; perimeter
G1 X106.039 Y106.039 E0.60663 ; perimeter
G1 X123.961 Y106.039 E0.60663 ; perimeter
G1 X123.961 Y123.901 E0.60460 ; perimeter
M204 S1000 ; adjust acceleration
G1 X124.368 Y124.368 F7800.000 ; move to first perimeter point
M204 S800 ; adjust acceleration
G1 F2700.000
```
Photos : 

[image]

PS : 
[image]

Thank you

rtyr, 2021-05-16:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/48a93e40fb2499832ea4dd61185c58912f6ef18a. See https://github.com/prusa3d/PrusaSlicer/issues/6362#issuecomment-820468245. 

Closing.

closed 2021-05-16 by rtyr

Zazou89, 2021-05-16:
Hey,

Thank you, I compiled PS (Version 2.4.0-alpha0), unfortunately same behaviour.

rtyr, 2021-05-16:
Please provide your 3MF project file and the g-code.

Zazou89, 2021-05-16:
Sure : [xyzCalibration_Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/6490181/xyzCalibration_Cube.zip)

rtyr, 2021-05-17:
The g-code is OK, no redundant deretraction (which was present in 2.3.0).

[image]

M600 behavior is completely handled by the firmware.