prusa3d/PrusaSlicer #6524: cambio de filamento, a pesar que lo agrego no se resuelve

gmmanka, 2021-05-16:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

gmmanka, 2021-05-16:
al agregar un cambio de filamento y que el soft lo muestra, no envia la orden a la impresora para que esto ocurra y imprime todo sin pausas

rtyr, 2021-05-16:
> Google translate:
> When adding a filament change and the soft shows it, it does not send the order to the printer for this to happen and it prints everything without pauses

Please provide your 3MF project file, PrusaSlicer version, OS, ....

gmmanka, 2021-05-16:
todos los archivos no se pausan en el cambio de color, version 2.3.0, windows 10 pro 64bits

rtyr, 2021-05-17:
Please provide your 3MF project so we can see whether the g-code is OK. Color change might be not supported on your printer (what is your printer?).

rtyr, 2021-06-30:
No response. Closing.

closed 2021-06-30 by rtyr