prusa3d/PrusaSlicer #6526: Avoid crossing perimeters & thin models

Moyrn, 2021-05-18:
### Version
2.3.1

### Operating system type + version
Win10 18363.1556

### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S

### Behavior
Printing a screw with empty cylinder inside.
So the screwing part is very thin : only one perimeter int + one perimeter ext
> the screwing part does not print correctly if at all. Under extrusion

i tend to use  a "fit for all" profile because i print many different pieces, so i do not want to have tons of profiles.
in this context, i would not need the "avoid perimeters", but it is enabled for my other purposes
(i use silk pla, scars are very visible, so i disable retractation... so i need avoid perimeters)

I initially though about clogged extruder, or too loose, or heat...
i played with standard profiles for printer, filament, and print. It was printing fine. so i step by step put back my settings, to finally - or so it humbly seems to me - that the issue comes from "avoid perimeters" setting - which is disabled by default in the standard settings.

* _Expected Results_
normal print of 2 perimeters

* _Actual Results_
under extrusion
i found that ticking "detect thin walls" did not prevent the issue
but if i disable "avoid crossing perimeters" > suddenly it works
so i guess avoiding crossing perimeters when there are ONLY perimters should be "auto disabled" or something like this

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[CBUIS12-PS.zip](https://github.com/prusa3d/PrusaSlicer/files/6498869/CBUIS12-PS.zip)

avoid perimeters is ticked by default. untick to solve the issue.
the bottom part is printing fine. only the thin screw section is heavily underextruding

rtyr, 2021-05-18:
It is likely just a result of longer travel move (more time for the filament to ooze from the nozzle), especially when the retractions are completely disabled in your settings.

Moyrn, 2021-05-18:
i took a photo (recovering the faulty parts from the trashcan
[image]
)

rtyr, 2021-05-18:
It seems like heat creep issue. Does it happen if you disable the ironing?

Moyrn, 2021-05-18:
yes. I tried it. I also thought about that. I faced heat creep a lot
in the past (see my comment below)
Printed at 205 if i remember well. In my garage, in a box. I have a
digital thermometer indicating T°C. Globally between 25 and 32°c in
the box these days. Not more

But keep in mind that even with ironing, it works when avoid
perimeters is disabled.

And secondly, I moved to BNBSX extruder to mitigate heat creep. I
print with fine details and slowly, and often with SILK PLA, which is
soooo soft (a real pain), so heat creep with std extruder (engine hot
>> axis >> bondtech gears hot) was constantly grinding and
underextruding/failing.

But here this filament is a filamentum trafic black.
The silk PLA is to brittle for printing "usable" thin pieces. Of
course I will move to PET or TPU soon. Only prototyping this piece
yet.

Thanks for your kind assistance.

Moyrn, 2021-05-18:
then be at ease. i found a way out. i'm not asking for help nor recognition ;-)
I'm just reporting something on which I spent more than one day to
find out something simple that works (untick a box)
I can understand that "avoiding perimeters" when there are "only
perimeters" to print is a tricky case.
if you find it usable fine, if not then i'm ok to close.

rtyr, 2021-05-18:
I am printing it on stock MK3, so we will see.

rtyr, 2021-05-18:
It finished without issues (sliced with your project file).

[image]

I think it is almost certainly heat creep issue. The ironing is at the same level, where your issues began. Ironing uses only 15% of the flow relative to normal object's layer height (which is already low at 0.1mm). It is not uncommon issue with ironing + PLA (especially during hot days or with enclosure), the flow is very small. It is mainly about the hotend and its cooling (and the filament of course). You also mentioned grinded filament at the Bondtech level, which IMHO confirms this theory.

Moyrn, 2021-05-18:
ok ! thanks for the investigation. Pretty neat.
Only weird aspect is the "avoid perimter" effect

perhaps that it slows the print after the ironing (although i had the same
pb without ironing)
no matter... the filament progresses slower and is ground

perhaps simply printing faster would do it

many thanks for your kind support.
shall i close ?

Wilfried

rtyr, 2021-05-18:
Yes, I think we can close this.

closed 2021-05-18 by Moyrn