prusa3d/PrusaSlicer #6527: PrusaSlicer 2.3.1   Dotaz na profil pro tiskásnu Creality CR-6-SE,

petr488, 2021-05-18:
PrPrusaSlicer 2.3.1, W10
CR-6-SE nefunguje s Průša slacerem, i když v programu profil je ale nefunguje, neproběhne inicializace- vyška Z a ani počátek xy, Zkoušel jsem porovnávat profily v Průša slaceru v funkčním Cura slaceru.  V Průšovi je jiná hodnota rozměrů desky, dál se to shoduje a nějaké hodnoty nejde porovnat, ale co je úplně jinak je Start a End G-code. Když překopíruji  Start a End G-code z Cury do Průši, opravím x,y  z Cury  a ještě je tam pár hodnot, které neznám může to chodit ?

Mate s tím nějaké zkušenosti?
Kde se dá stáhnou funkční profil pro tiskásnu Creality CR-6-SE? Nebo co se má vyplnit v profilu, aby to chodilo?
Děkuji  Petr

rtyr, 2021-05-18:
Máte verzi nastavení 0.0.16?

[image]

petr488, 2021-05-19:
Mám firmware ver. 1.0.3.6
print size 235x235x250.

rtyr, 2021-05-19:
Myslím verzi nastavení PrusaSliceru, viz obrázek.

petr488, 2021-05-19:
> Velikost tiskové plochy je tak jak je záměrně. Autor profilů to zdůvodnil
> tak, že tím tak zamezí potencionální kolizi modelu s úvodní purge linkou.
>
> Teď jde spíš o to, jestli máte poslední nastavení pro Creality (0.0.16,
> verzi najdete viz obrázek). Podobný problém jako vy hlásil i jiný člověk,
> na základě čehož se dělala drobná úprava ve startovacím g-codu. U daného
> člověka se ale nakonec ukázalo, že měl vadnou desku, což je zjevně nějaká
> známá častá závada na CR-6SE.

Ano, mám tu starší verzi 0.0.15 a to  jsem ten program instaloval asi před
14 dny.  Mám přeinstalovat celý program nebo se dá někde stáhnout profil?

Moc děkuji za odpovědi, Průša Slicer je velice přívětivý a v češtině.

rtyr, 2021-05-19:
Po spuštění PrusaSliceru byste měl vidět vpravo dole notifikaci upozorňující na dostupnou aktualizaci nastavení, stačí na ní kliknout. 
Daný počítač musí být připojený na internet a v parametrech programu musí být povolená aktualizace profilů (to je zapnuté standardně). Ještě můžete zkusit volbu pro kontrolu aktualizací (menu konfigurace).

rtyr, 2021-05-21:
Podařilo se?

petr488, 2021-05-21:
Včera se program aktualizoval a hned jsem si udělal tiskové soubory. Ale
teď v práci jsem se k tisku ještě nedostal. V příbalu tiskárny bylo
upozornění na

G28 X0Y0
G28 Z0         nesmí být   jen G28    a dál šlo G92

no a to tam asi je upravené.

Moc díky. dnes to ještě  prubnu. PP

rtyr, 2021-05-21:
Dejte prosím vědět, jestli spuštění tisku proběhlo úspěšně. Jen ještě upozorním, že je tam trochu nestandardní spuštění tisku, že se tryska v první fázi nahřívá na 120C, to je záměr autora proflů.

To upozornění v příbalu tiskárny je zajímavé. V profilech od Creality v Cuře je samotné G28. :-) Uvidíme, jestli vám to bude fungovat tak jak to je teď. Případně to pak můžeme předělat na ten nově doporučovaný tvar. Mám trochu obavy, že existují různé verze stejné tiskárny a že výrobce zápasí s firmware.

rtyr, 2021-05-21:
@pmjdebruijn FYI. Creality included some sort of warning note with the printer (CR6SE), stating that start g-code should include `G28 X0 Y0`, then `G28 Z0`, instead of traditional `G28`. Petr will try current start g-code (previous start g-code from 0.0.15 didn't work), so we will see if it works or not.

pmjdebruijn, 2021-05-21:
@rtyr, specifically X0 and Y0 and Z0? that's very non standard isn't it? That will probably break for firmware for other Creality printers or stock marlin? And that would force us to have `start_gcode` that's entirely CR6SE specific, which would be rather miserable...

But the latest firmware seems to be V2.0.1.3, not V1.0.3.6, so presumably Creality may have fixed this? Updating the printer firmware to the latest seems like the better solution...

petr488, 2021-05-21:
> I just look in the 1.0.3.6 firmware sources for G28, but the latest
> doesn't seem to look for X0 or Y0 or Z0, unless I've missed something at a
> glance... So possibly G28 X Y followed by G28 Z might work just as well,
> please ask @petr488 <https://github.com/petr488> to test that as well...

Spuštění tisku proběhlo dobře. Tiskárna správně najede do prostřed desky a
nastaví výšku trysky nad deskou odjede do počátku souřadnic a udělá první
čáru a začne tisknout a vykreslí model. Problém mě ale nastal z výškou
trisky nad dekou. Už první čára prakticky nedržela a byla to kulatá nit.
Nastavená výška 0,2mm. Tisk se zlepšil při nastavení na 0,1mm.
Tiskárnu jsem potom nakalibroval desku při 120 st. a potom s nahřátou
tryskou výšku trysky nad deskou na 0,2mm. Důkladně jsem očistil špičku od
filamentu a nechal vychladnout. Spustil jsem tisk a všechno proběhlo OK.
Myslím, že je problém s nastavením výšky trysky při 120 st. Z trysky pořád
teče troška filamentu, takže jak triska zchladne na 120 st.  vždy je tam
nějaký zbytek filamentu a asi to je problém špatného měření výšky trysky
nad deskou. Ani nevím jestli to dělám dobře, ale kalibruji horkou trysku na
teplotu filamentu. Tryska sjede dolů filament se otiskne na desku triska
uhne kousek na horu zastaví se a pak se nastaví výška a když chcete
zkontrolovat protažením papíru výšku mezery, tak to nejde, protože je tam
rozplácnutý zbytek filamentu. Mně se to moc nelíbí.
Začátek tisku při teplotě 120 st. by pak byl - vysunout zpět filament,
zahřát trysku a očistit, nechat zchladnout, vrátit filament a pak
spustit tisk a to pokaždé. Já to ještě vyzkouším.   Moc děkuji za pomoc.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_-4390361601435279725_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rtyr, 2021-05-21:
@petr488 Díky za info. Těch 120C se tam dávalo právě kvůli této tiskárně (Creality to tak zezačátku mělo v oficiálních profilech pokud vím). Osobně bych tu teplotu zvýšil na cca 170C, což by stále mělo zabránit většímu samovolnému vytékání filamentu, ale zároveň by filament byl dostatečně měkký, aby nezpůsoboval problémy při homování.

@pmjdebruijn 
So it works fine with G28, but it seems that 120C is not good for probing/Z homing (it will trigger too soon because of hard filament sticking from the nozzle). I suspected that it could be problem given how 6SE probing works. I don't see any reason to use 120C. If I remember correctly, you set this temperature globally because of CR6SE (it was probably recommended by Creality at the time). But it seems that it only causes issues on that printer. I guess we can globally increase the temperature to something like 160-170C, it should still prevent oozing of the filament (while Z homing on 6SE should be more reliable with higher temp).

petr488, 2021-05-22:
Dá se ta teplota 12o st. změnit a nastavit těch 170st. Rád bych to zkusil,
myslím si, že to moc pomůže.  Stačí přepsat v G start - kodu :  M104 S120
na  S170 - nebo je to daleko složitější. Díky.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

pmjdebruijn, 2021-05-22:
@petr488 can you test 150C as well? as I'd highly prefer that over 170C...

rtyr, 2021-05-22:
> Stačí přepsat v G start - kodu :  M104 S120 na  S170 - nebo je to daleko složitější.

Stačí.

petr488, 2021-05-24:
Dnes jsem vyzkoušel jak startují programy Creality Slicer 4.2.1. dodaný s
tiskárnou (je to Cura ) a Ultimaker Cura 4.8.0 zkusil jsem několik modelů,
tisk startoval u obou programů stejně - nahřála se deska na teplotu pro
filament po dosažení teploty skočí teplota trysky  z 0 na nastavenou
teplotu pro tisk filamentu. Nahřátí na nějakou nižší teplotu oba programy
přeskakují. Tiskárna dosáhne nastavenou teplotu otestuje se výška
první vrstvy a hned se tiskne.
Pak jsem chtěl vyzkoušet Průša Slicer a tak jsem si ho v práci
nainstaloval, ale verze 0.0.15 se  mně nepodařilo aktualizovat, tak jsem si
myslel, že je to možná už upravené. Ale chyba lávky, tiskárna vyhodila
error a jela dál a pak začala zase hrkat, nevím jestli to přeskakuje
motor nebo  řemenice ale zní to hrozně a asi to dělá pokaždé, vrhl jsem se
na vypínač... Mám firmware ver. 1.0.3.6.
Dá se hned  přejít na teplotu filamentu?   Moc děkuji.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rtyr, 2021-05-24:
Nejsem si jistý, že rozumím. Poslední verze nastavení pro Creality je 0.0.16, s 0.0.15 vám to asi nebude fungovat. Stačí aspoň 2x spustit PrusaSlicer a měla by se nabídnout aktualizace. 

To s tím snížením teploty je tak globálně u všech Creality tiskáren v PrusaSliceru. Je to trochu nestandardní, ale nemělo by to způsobovat žádný problém. Autor profilu pmjdebruijn navrhoval zvýšení teploty ze 120C na 150C.

Jinak asi by bylo vhodné aktualizovat i firmware tiskárny.

petr488, 2021-05-24:
Nainstaloval jsem Prusa Slicer verze 0.0.15 a aspoň dvakrát jsem
nastartoval a ještě restartoval ale verze 0.0.15 zůstala a když jsem se
podíval na aktualizace , tak že není žádná dostupná. Zkusím to zítra.
a stou nižší teplotou to není problém. Díky moc.
Já žiju jen z vašich manuálů ( tuneluju ) a tak malou poznámku na
čištění trysky.  Měl jsem troch ucpanou trysku a občas kopal podavač či
extrudér. Při čištění jsem zjistil že je dobré zahřát trysku na plno 250st
vytáhnout teflonovou hadici a do díry zastrčit filament až do trysky a teď
co je navíc : vzít drát na čištění trysky a zasunout do trysky ze spodu a
teď ze spodu tlačit drát a z hora táhnout filament až se vytáhne špunt
filamentu z trysky ven a potom se zpět dolů vytáhne drát. Kolem drátu
filament zchladne a tahne se nahoru ven. Šlo to lépe než postup co mate na
stránkách. Pokud to je k něčemu , tak to prosím přepošte.  Díky

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

petr488, 2021-05-25:
Dobrý den,  nepodařilo se mně program aktualizovat, 5x znovuspuštění i
restart PC.  Díky

rtyr, 2021-05-25:
Zkuste volbu "Zkontrolovat aktualizace" v menu Konfigurace.

Pokud ani to nepomůže, znamená to pravděpodobně, že dané PC z nějakého důvodu nevidí na náš server (blokováno např. vaším antivirem, firewallem).

Jinak aktualizace se neprovádí automaticky, je třeba kliknout vpravo dole na notifikaci.

petr488, 2021-05-25:
Přepojil jsem se na veřejné WiFi a už aktualizace prošla. Díky za pomoc.

petr488, 2021-05-28:
Pár dnů jsem *PrusaSlicer používal a funguje to takto : na tiskárně
*(CR6SE)* se
spustí tisk ze souboru, tiskárna  pomalu najede na střed a je jedno jaká je
nastavena teplota na ni nečeká   a " naváží " výšku trysky  dál najede do
počátku*
* a naskočí teplota filamentu, po dosažení se rozjede tisk.*
*Já startuji - předehřeji na PETG a spustím tisk ....  To se dá
rutyně používat. OK*

CR6SE využívá pro kalibraci váhové čidlo a moc záleží na tom v jakém stavu
je špička trysky, teplota, zbytky filamentu...

*Na výtisku z PETG svislých stěn se separují obvodové perimetry, celý se
sloupne. Zvýšil jsem násobič extruze na 1,1 a zdá se to dobré.*
*Dá se tato separace odstranit  nastavením ještě jiných parametrů než *násobič
extruze.   Děkuji

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rtyr, 2021-05-28:
@pmjdebruijn I checked the start g-code based on the feedback from Petr and I noticed there is only M104 S120 and 10s delay (no M109). It is certainly not good approach for CR-6SE (given how probing works there). The temperature needs to be higher and must be reached **before** homing.

pmjdebruijn, 2021-05-28:
I'm fine with raising 150C in any case, given the 120C was initially introduced for the CR6 G29, possibly this was only valid for old firmwares?

I'll look into what we can do with the start_gcode over the weekend, as I want something that is at least vaguely elegant and maintainable into the future...

rtyr, 2021-05-28:
@petr488 Díky za info. Do startovacího g-codu by se mělo určitě přidat čekání na dosažení nějaké rozumné teploty před homováním osy Z (teď je tam nastavení na 120C, ale na dosažení teploty se nečeká, je tam jenom 10s pauza). Vzhledem k tomu, jak CR6SE v tomto ohledu funguje určitě není žádoucí homovat se studenou tryskou.

Pokud jde o váš druhý problém, tak to se obávám může mít víc příčin. Pokud tiskárna celkově podextrudovává, bude pravděpodobně nesprávně zkalibrovaný extruder (špatně nastavený počet kroků motoru/mm ve firmware, v případě Creality zřejmě dané nestabilní/nepřesnou velikostí podávacího kolečka při výrobě - myšleno ve smyslu, že je různý průměr na jednotlivých kusech tiskáren). Taky může být problém v příliš nízké teplotě pro daný filament, přiucpané trysce, apod. Můžete taky zkusit to samé vyslicovat v jejich sliceru s jejich profilem pro potvrzení.

rtyr, 2021-05-28:
> 
> 
> I'm fine with raising 150C in any case, given the 120C was initially introduced for the CR6 G29, possibly this was only valid for old firmwares?
> 
Who knows. They are using only G28 now (no G29, ....and there is some note to not use G28 while they are using it in their own profiles). It is possible they initially wanted to prevent oozing during MBL, but then realized it affects the reliability of the probing too much (120C is not enough to prevent false probing).

petr488, 2021-05-29:
Děkuji za tipy, já to ještě vyzkouším a porovnám s Curou. Připadá mně to,
že se triska kontinuálně pomalu ucpává. Teplotu mám pro PETG 240 až 245st.
Funguje čistící filament? S PLA jsem problémy neměl ale s PETG ano.
V jakém rozsahu je možné nastavovat nasobitel extruze 0,9 až 1,1 nebo je to
ještě víc?
Dá se podle nějakého parametru extruze porovnat jednotlivé profily
filamentů.     Moc Díky.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

pmjdebruijn, 2021-05-30:
@rtyr: I've just filed #6572, where nozzle temp is raised to first_layer_temp-50 before moving onto G28, for the CR6 series of printers...

rtyr, 2021-06-01:
> Připadá mně to, že se tryska kontinuálně pomalu ucpává.

To může mít více příčin. Pokud máte podezření na přiucpanou trysku, je vzhledem k ceně nové asi nejjednodušší začít její výměnou. Jinak můžete zkusit čištění metodou https://help.prusa3d.com/cs/article/cold-pull-mk3s-mk2-5s_2075/. Problém může být i na straně extruderu, zkontrolujte, jestli je správně nastavený přítlak (ani málo ani moc). Toto je spíš na technickou podporu Creality :-).

Byla vypuštěna nová verze nastavení pro Creality 0.0.17. Je tam upravený startovací g-code takovým způsobem, že se před homováním čeká na dosažení teploty `Cílová teplota - 50C`. Pokud má filament pro první vrstvu nastaveno např. 210C, předehřeje se tryska před prvním homováním na 160C (210-50). Také se před homováním čeká na úplný předehřev podložky. To by mělo řešit zmíněné problémy při startu tisku. Zavírám issue.

closed 2021-06-01 by rtyr

petr488, 2021-06-03:
Nová verze nastavení pro Creality 0.0.17.  Aktualizoval jsem program a
start tisku funguje skvěle, tak jak jste psal. Těch - 50st. je asi akorát.
Tiskl jsem PETG, skočilo 190 st. a  čekalo se na dosažení teploty, jak se
rozběhla tiskárna, očistil jsem trysku a tiskárna nastavila výšku trysky a
tiskárna najela do počátku a naskočila teplota filamentu a jak se toho
dosáhlo rozjel se tisk. OK spokojenost.  Děkuji.

Prosím, má násobitel extruze nějaké hranice ? Může extruzi také ovlivnit
změna  šířky extruze ?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Bez
virů. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rtyr, 2021-06-03:
> Prosím, má násobitel extruze nějaké hranice ? Může extruzi také ovlivnit
> změna  šířky extruze ?

Omezený není, ale pokud je tiskárna v pořádku a je správně zkalibrovaný extruder, měl byste se u běžných materiálů pohybovat kolem hodnoty 1 (dejme tomu +- cca 10%).  Šířka extruze je něco jiného než násobič. Pokud tiskárna podextrudovává, zvětšení šířky extruze není řešení.