prusa3d/PrusaSlicer #6528: Slice Problem of filling inside cyclinder

FR86YoPPZ, 2021-05-18:
### Version
_Version of PrusaSlicer used goes here_
*2.3.1.+win64**

### Operating system type + version 
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### Behavior
* _Describe the problem_
* When i have a cyclinder, the filling inside is not correct. There is a gap between wall and filling. check screenshot.
* I don't remenber to have meet this problem in previous version ?!
* 

* _Steps needed to reproduce the problem_
[Golf_Epuisette_Blocage_Fixation_v2_pbl.zip](https://github.com/prusa3d/PrusaSlicer/files/6504261/Golf_Epuisette_Blocage_Fixation_v2_pbl.zip)

Place the part attached in zip horizontally and press slice. You will see gap in cylinder. Bad slice inside cyclinder.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
[image]
[image]

FR86YoPPZ, 2021-05-18:
my problem comes from from an option ticked. "filling only if necessary"

closed 2021-05-18 by FR86YoPPZ

rtyr, 2021-05-18:
It is caused by the "Only infill where needed" option. Disable it.