prusa3d/PrusaSlicer #6530: Issues slicing  a rectangular grid

cezar1, 2021-05-19:
### Version
2.3.1

### Operating system type + version
Windows 10 64 bit 21H1

### 3D printer brand / version + firmware version (if known)
Mendel90 with Marlin firmware
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
The slice output is wrong.
Import the file in slicer
Slice it

#### Project File (.3MF) where problem occurs
[image]

[WindowSlit.zip](https://github.com/prusa3d/PrusaSlicer/files/6505933/WindowSlit.zip)

neophyl, 2021-05-19:
Sliced fine for me once I turned off Detect thin walls.

cezar1, 2021-05-19:
Yes it works, thanks sir!

closed 2021-05-19 by cezar1

rtyr, 2021-05-19:
I was able to reproduce it in 2.3.1, but not on current master.