prusa3d/PrusaSlicer #6531: Support inside Object in interlocked Objects

swissbear, 2021-05-19:
### Version
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+

### Behavior
I have 3 interlocking objects
[image]
Obviously its important how these objects are aranged, as they have to be interlocked.
When generating supports from build plate only, the supports go straight through the objects:
[image]

#### Project File (.3MF) where problem occurs
[Kreuzknoten_support_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/6509972/Kreuzknoten_support_bug.zip)

bubnikv, 2021-05-19:
You have to slice them as a single object. Select them on plater, click
right mouse button to show a pop-up, select "merge objects".

swissbear, 2021-05-19:
Thank you for this quick reply
Applying this, there will still be a bit of support generated ON the object...

rtyr, 2021-05-19:
> Applying this, there will still be a bit of support generated ON the object.

Support generation was significantly improved for PrusaSlicer 2.4.0. It is ok on current master.