prusa3d/PrusaSlicer #6532: Brim should start from the inside

WestleyR, 2021-05-20:
### Version
2.3.1+x64

### Operating system type + version
macOS Big Sur 11.3.1

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ (latest firmware as of this issue)

### Behavior

Brim should start from the inside, and work its way out. Otherwise, the over extruded first layer develops an error.

[image]

[image]

Alternatively, the brim rings should alternate, and then fill the gap. It should also do this when infilling the first layer (to avoid PETG "balling"), but thats another issue.

#### Project File (.3MF) where problem occurs

Any object with a brim > 3mm.

rtyr, 2021-05-20:
The issue on the photo is caused by incorrect live Z setting, the nozzle is too low.

WestleyR, 2021-05-21:
Yes, in most cases raising the nozzle to the "correct" hight would fix that issue. However most of the filaments I print with require a lower first layer, to further improve bed adhesion.

However I see no issue starting the brim from the inside, that would definitely help this issue.

closed 2026-09-29 by prusaslicer-bot[bot]