prusa3d/PrusaSlicer #6533: Slicer does not respect settings at all

proc, 2021-05-20:
### Version
2.31
### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
snapmaker 2

### Behavior
speed, infill, layer height, support settings are not respected unless you right click the object and add custom settings

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
any project

rtyr, 2021-05-20:
Please provide your 3MF project file.

proc, 2021-05-20:
Attached. I can also make a video of this occurring if you cannot reproduce. This is also happening 100% of the time on 2.3.0 as well.

neophyl, 2021-05-20:
Your attached project already has the per object settings which over ride the print profile settings.  Changing the % infill in the Object panel changes the infill % as expected.  
If I remove the the per object infill setting, then changing the global print profile % Infill from the top dropdown (located top tight below the supports option dropdown) works correctly and does modify the infill on your Benchy.  So the program is working normally for me.  Checked using 2.3.1 on Windows 10.

proc, 2021-05-20:
> Your attached project already has the per object settings which over ride the print profile settings. Changing the % infill in the Object panel changes the infill % as expected.
> If I remove the the [...]

Interesting. It doesn't work for me. *shrug*

neophyl, 2021-05-20:
For me changing the global infill dropdown setting does result in the export gcode button changing to Slice Now.  Even if I have the per object infill % over ride still there.  Changing either of the infill dropdowns updates that button.  

Changing the global settings doesn't invalidate the slice though when the object has its own infill over ride.  By that I mean if you slice and are on the preview tab normally and you change a setting the plater is cleared as the slice preview is no longer valid.  When the global setting is changed it updates the button but doesn't clear the slice.  If I update the object infill over setting then it does clear the plater as the slice is invalidated.

If I remove the object over ride then changing the global setting does invalidate the slice and clears the preview.

rtyr, 2021-05-20:
Just remove per object settings (blue arrows) if you want global settings to take effect. Per object settings overrides global settings.

[image]

I believe we can close this issue.

proc, 2021-05-20:
that doesn't fix it

proc, 2021-05-20:
it doesn't work with or without variable layer height, and the cooling i've changed every single print to try and debug.

proc, 2021-05-20:
your project works, but my default project doesn't here look at this one I saved it in a broken state. I have infill set to 80% in the top menu but the preview is only showing 15%, it did not trigger the "slice now". Even when I hit "reslice now" it doesn't update the preview. There are no object settings. What could cause that? 
[3DBenchy (2).zip](https://github.com/prusa3d/PrusaSlicer/files/6514827/3DBenchy.2.zip)

neophyl, 2021-05-20:
Can I ask where you got your printer profile from ?  I dont know if that printer is bundled as part of PS or if you downloaded it from a website somewhere.

The reason I ask is that over on the Prusa Forum we had a case where there were a bunch of settings that no matter how the user altered them the results didn't change, which seems to be what you are experiencing.

That users problems stemmed from the fact they had imported a profile for their printer from somewhere online.  The profile for the Printer also contained setting keys that were normally in the Print and Filament profiles.  This meant that when they edited the keys in the print and filament profile they would change in the gui and look ok but when it was sliced it used the duplicate keys from the Printer profile which of course could never be edited as the Printer profile doesn't have those values in.  

It was difficult to track down as importing their 3mf worked as expected for other people.

I'm not saying that is what is occurring here but its at least a possibility I think.

proc, 2021-05-20:
These profiles are copied from the profiles shipped with prusaslicer, I just detached them and copied them

neophyl, 2021-05-20:
Well that rules out that possibility then :)

proc, 2021-05-20:
yea I think you were right, I created a custom printer and right off the bat the settings are working again. Thanks :)

neophyl, 2021-05-20:
Just found the thread too https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-layer-height-always-0-2/  and it was a snapmaker.

proc, 2021-05-20:
bingo

neophyl, 2021-05-20:
In which case I think this needs to become a feature request such that when importing profiles via a File >Import>config/from project/bundle  PS should check for existing defined keys that belong in other sections and only load in the keys that belong in that section.

That should stop issues with users importing in malformed configuration files they download.

rtyr, 2021-05-20:
That's interesting. Thanks @neophyl.

rtyr, 2021-05-20:
Incorrect prusaslicer bundle mentioned above:
[1613551274-demoForum.zip](https://github.com/prusa3d/PrusaSlicer/files/6515530/1613551274-demoForum.zip)

rtyr, 2021-05-20:
I tried to import that prusaslicer_config_bundle.ini to the PS2.3.1 and incorrect keys were automatically filtered during the import.

@proc Can you share the link, where you downloaded your profile?

rtyr, 2021-05-20:
It seems something went wrong during the upload.

proc, 2021-05-20:
sorry let me try again 
[PS_SM2.zip](https://github.com/prusa3d/PrusaSlicer/files/6518259/PS_SM2.zip)

rtyr, 2021-06-01:
In the current PS2.3.1, the incorrect/duplicit keys are automatically filtered during the import of problematic bundle like https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-layer-height-always-0-2/. Problematic bundle was likely imported by some old version of PrusaSlicer or printer profile was modified manually in data directory. Closing.

closed 2021-06-01 by rtyr