prusa3d/PrusaSlicer #6534: Bug: unexpected filament unload during single filament print

QuInsight, 2021-05-20:
PrusaSlicer 2.3.0
Windows 10 Home
Prusa i3 MK3
Firmware 3.9.1-3918
### Behavior
For several months I have been getting random stops for filament unload / reload on single color single filament prints. I cannot deliberately reproduce the problem, but I have attached a zipped gcode that gave me the problem at around 17 minutes into a 20 minute print. I have reviewed the gcode in Prusa g-code Viewer but I don't see any evidence of a filament switch in the file. 
[Unexpected Filament Change.zip](https://github.com/prusa3d/PrusaSlicer/files/6512870/Unexpected.Filament.Change.zip)

Re-slicing and saving new gcode seems to fix the problem. 

_Is this a new feature request?_ NO

I thought I could live with the problem until it started occurring multiple times on an 18 hour print. It always seems to happen near the end of the print too, so in that case I had a lot of material and time invested and I had to tend to the printer about every 5 minutes over the final two hours of printing.

rtyr, 2021-05-20:
There is no filament change in the provided g-code. The issue is likely caused by filament sensor (false filament runouts), you can disable filament sensor in LCD menu.

QuInsight, 2021-05-20:
Thank you for the speedy answer! I'll temporarily disable sensor and work on getting a new one if that fixes the problem. I need a working filament sensor!

QuInsight, 2021-05-20:
LOOSE CABLE! Thanks.

closed 2021-05-20 by QuInsight