prusa3d/PrusaSlicer #6535: Don't support bridges can not be disable in command line

Mpolo10, 2021-05-20:
### Version
Version 2.3.1

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
We can't disable the 'dont-support-bridges' option when we run PrusaSlicer in command line.

'dont-support-bridges' option is set 'true' by default and the option accepts no paramater so we can not turn it off.
That is to say of we add '--dont-support-bridges' we can see in the Gcode 'dont-support-bridges = 1' (means true) and if we do not add the option, the Gcode generated displays 'dont-support-bridges = 1' too.

When I run the PrusaSlicer GUI i can disable dont-support-bridges and it does the job.

_Is this a new feature request?_
more or less

combolek, 2021-05-23:
To disable boolean option you have to use the `no-` prefix like this:
```bash
$ prusa-slicer --slice --no-dont-support-bridges bridge-Body.3mf 
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
85 => Generating support material
88 => Generating skirt
90 => Exporting G-code to bridge-Body_0.2mm_PLA_MK3S_{print_time}.gcode
Slicing result exported to bridge-Body_0.2mm_PLA_MK3S_1h59m.gcode

$ grep dont_support_bridges bridge-Body_0.2mm_PLA_MK3S_1h59m.gcode
; dont_support_bridges = 0
```
Yeah, I know, but it works :slightly_smiling_face:

[bridge-Body.zip](https://github.com/prusa3d/PrusaSlicer/files/6527343/bridge-Body.zip)

Mpolo10, 2021-05-25:
Thank you ! Indeed it seems to do the job 😉