prusa3d/PrusaSlicer #6539: Feature: seam position - inside

KubaMiszcz, 2021-05-21:
add seam position mode where:
it tries to find holes/internal areas and place seams on internal perimeters

rtyr, 2021-05-21:
I don't understand your request. Please provide more detailed description, screenshot, etc. Do you mean internal areas (from the object's POV) of the external perimeters?

KubaMiszcz, 2021-05-21:
edited

3Delights, 2021-05-21:
They probably can't due to Stratasys patenting the concept (like so much other stuff). See here:
[Patent via Google](https://patents.google.com/patent/US8349239?oq=8349239).
They are really stifling innovation.

KubaMiszcz, 2021-05-21:
really? this way I can patent whole idea of slicing objects at all

SergeyMokhov, 2022-04-09:
> 

If that was true,[ thin wall setting](https://patents.google.com/patent/EP2176805B1/en?inventor=Donald+J.+Holzwarth) would not be allowed as well

SergeyMokhov, 2022-04-09:
> Please provide more detailed description, screenshot, etc
I think original author meant something like this
[image]

KubaMiszcz, 2022-04-09:
@SergeyMokhov  yes, exactly what you posted, 
I mean that at the end when print is ready, all seams are inside, not on external surfaces, of course when its possible and internal areas exist

just additional condition above random/aligned/rear

bubnikv, 2022-06-29:
As long as the perimeter lines are not allowed to cross, there will always be a seam on the outside.
Your suggestion cannot be implemented.

closed 2022-06-29 by bubnikv

SergeyMokhov, 2022-06-29:
Maybe this could work only if perimeters are allowed to cross? You know, when you check one option a popup window shows to confirm you want to allow perimeters to be crossed?

KubaMiszcz, 2022-06-29:
sure, tbh I didnt tested crossing perimeters and how it affects prints