prusa3d/PrusaSlicer #6544: Skirt-brim collision

tomu4, 2021-05-22:
### Version
_2.3.1+win64_

### Operating system type + version
_Windows 10 Pro, version 20H2_

### 3D printer brand / version + firmware version (if known)
_not apliable - slicer issue_

### Behavior
* _When brim button is ticked (I'm using tickbox on right side of main screen, marked by blue color on screenshot below), skirt should be either disabled, or skirt_distance should be automatically set to bigger distance to prevent skirt-brim collision_ [image]
* _Tick brim checkbox on main (preview) screen, than press Slice._
* _Expected Results: Skirt is disabled or moved further away from object_
* _Actual Results: Skirt and brim take same position - skirt usually printed first and than brim is printed over it._
  * _Screenshots from __*PrusaSlicer*__ preview: [image]_

_Is this a new feature request? Yes, it is._

#### Project File (.3MF) where problem occurs
_3MF file is not supported on github, but this issue should appear with all project files. Image documenting issue is attached in text above._

rtyr, 2021-05-22:
Why this behavior was implemented:
#724

Duplicate issue + links to other similar issues:
#3692

Closing.

closed 2021-05-22 by rtyr

tomu4, 2021-05-22:
@rtyr I understand potential use of this behavior, however, I think that if you only use 3 layers of skirt (default), you usually don't want (or at least don't need) skirt to touch brim. I understand there can be certain situations where this behavior can be helpful, but I think in majority of cases, you go to settings and change it manually anyway. 

What I personally think could be more helpful and would make work easier (especially for new users with default settings), would be automatically increase skirt distance, but instead of setting zero next to brim like said in #724, leave zero at object and let user decrease it, if he has reason for it and wants to do so.

Other possible behavior I can think of could be: Detect where brim is intersecting with skirt and don't print first layer of skirt there, as there is already brim in that place. There will be no collision there, and printer will not print twice in the same place on the same layer. 

This said, I don't care which option stays in slicer, I just thought that this collision at the first layer could confuse newbies, especially if they don't go very well with technology.

rtyr, 2021-05-22:
We have some ideas how to improve it, so it may change in the future.

lukasmatena, 2021-08-12:
It has indeed been improved for upcoming 2.4.0-alpha1: https://github.com/prusa3d/PrusaSlicer/issues/4802#issuecomment-897492086