prusa3d/PrusaSlicer #6547: Color Button and Extruder dropdowns don't work on Manjaro KDE

Chrismettal, 2021-05-22:
### Operating system type + version
Operating System: Manjaro Linux
KDE Plasma Version: 5.21.5
KDE Frameworks Version: 5.82.0
Qt Version: 5.15.2
Kernel Version: 5.9.16-1-MANJARO
OS Type: 64-bit
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 2700X Eight-Core Processor
Memory: 31,4 GiB of RAM
Graphics Processor: GeForce RTX 2080 Ti/PCIe/SSE2

### PrusaSlicer System Info output
PrusaSlicer
Version:   2.3.1
Build:     PrusaSlicer-2.3.1+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.9.16-1-MANJARO x86_64
Total RAM size [MB]: 33,685MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.73.01
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2080 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA

(Installed from official arch repository)

### Behavior
On my system neither the color of individual extruders can be changed when clicking the orange boxes, nor can I assign an extruder to individual parts or modifiers with the dropdowns below, marked in red.

Pressing the orange box just opens the filament type dropdown, which I could live with since the color is only for preview. 
Pressing the extruder dopdowns below does not do anything so I can never choose anything but the default extruder. My workaround right now is firing up a Windows VM just to select the extruders, and then saving the .3mf, which is how I can see extruders 1 and 2 being selected in the screenshot.

[image]

#### Project File (.3MF) where problem occurs
Any multicolor project

lukasmatena, 2021-05-22:
What exactly is build PrusaSlicer-2.3.1+UNKNOWN and where did you get it? Does the same issue appear when running our AppImage (downloaded from https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1)?

Chrismettal, 2021-05-25:
2.3.1+UNKNOWN is just the one version that is provided in pamac in the Manjaro user repository:
[image]
Should just be release 2.3.1 I thought. 

I tried the appimage where I am in fact able to at least select an extruder (kind of fiddly, the dropdown isnt stable, seems to want to dissapear instantly, but at least it works) but not the displayed extruder colors, as they will always open the material dropdown and never the color.
With all versions, appimage or not I get locale errors like this, which I have to prevent by prefixing `LC_ALL=en_US.UTF-8` to my executables, which I haven't found yet for appimages:
```
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```
My locales are `en_US.UTF-8` for display language and `de_DE.UTF-8` for formats, but that's a different issue entirely, just a minor hiccup making appimages not as smooth to use for me.

I am trying to build PrusaSlicer myself to give more helpful information but I haven't figured out how to get past the following build errors yet (using static, independend deps):

```
[main] Building folder: PrusaSlicer 
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/chrismettal/GitLocal/PrusaSlicer/build --config Release --target all -j 18 --
[build] [6/123   0% :: 0.002] Running utility command for encoding-check-avrdude-slic3r
[build] [6/123   1% :: 0.007] Running utility command for encoding-check-semver
[build] [6/123   2% :: 0.050] Running utility command for encoding-check-avrdude
[build] [6/123   3% :: 0.107] Linking CXX static library src/libslic3r/liblibslic3r_cgal.a
[build] [6/123   4% :: 0.338] Running utility command for encoding-check-libslic3r_gui
[build] [6/123   4% :: 0.393] Running utility command for encoding-check-libslic3r
[build] [24/123   5% :: 9.936] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLShadersManager.cpp.o
[build] [25/123   6% :: 11.906] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoBase.cpp.o
[build] [26/123   7% :: 12.239] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoMove.cpp.o
[build] [27/123   8% :: 13.590] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoScale.cpp.o
[build] [28/123   8% :: 13.918] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLShader.cpp.o
[build] [29/123   9% :: 14.916] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoCut.cpp.o
[build] [30/123  10% :: 15.563] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/OpenGLManager.cpp.o
[build] [31/123  11% :: 17.776] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o
[build] FAILED: src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o 
[build] /usr/bin/c++ -DBOOST_ALL_NO_LIB -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_OPENEXR_STATICLIB -DOPENVDB_STATICLIB -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I../src -Isrc/platform -I../src/libslic3r -Isrc/libslic3r -I../src/libnest2d/include -I../src/miniz -I../src/glu-libtess/include -isystem ../src/eigen -isystem ../Deppo/usr/local/include -isystem ../src/libigl -isystem /usr/local/include/OpenEXR -Wall -Wno-reorder -O3 -DNDEBUG -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -DOPENVDB_ABI_VERSION_NUMBER=8 -std=gnu++17 -Winvalid-pch -include /home/chrismettal/GitLocal/PrusaSlicer/build/src/libslic3r/CMakeFiles/libslic3r.dir/cmake_pch.hxx -MD -MT src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o -c ../src/libslic3r/OpenVDBUtils.cpp
[build] In file included from ../Deppo/usr/local/include/openvdb/Types.h:34,
[build]                  from ../Deppo/usr/local/include/openvdb/openvdb.h:35,
[build]                  from ../src/libslic3r/OpenVDBUtils.hpp:12,
[build]                  from ../src/libslic3r/OpenVDBUtils.cpp:2:
[build] ../Deppo/usr/local/include/openvdb/version.h:84:10: error: #error expected OPENVDB_ABI_VERSION_NUMBER <= OPENVDB_LIBRARY_MAJOR VERSION_NUMBER
[build]    84 |         #error expected OPENVDB_ABI_VERSION_NUMBER <= OPENVDB_LIBRARY_MAJOR VERSION_NUMBER
[build]       |          ^~~~~
[build] [31/123  12% :: 18.390] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmosManager.cpp.o
[build] [31/123  13% :: 19.647] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/SLA/Hollowing.cpp.o
[build] FAILED: src/libslic3r/CMakeFiles/libslic3r.dir/SLA/Hollowing.cpp.o 
[build] /usr/bin/c++ -DBOOST_ALL_NO_LIB -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_OPENEXR_STATICLIB -DOPENVDB_STATICLIB -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I../src -Isrc/platform -I../src/libslic3r -Isrc/libslic3r -I../src/libnest2d/include -I../src/min [... cut ...]
[build] [31/123  22% :: 34.345] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoSlaSupports.cpp.o
[build] [31/123  23% :: 36.175] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Selection.cpp.o
[build] [31/123  24% :: 36.192] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLCanvas3D.cpp.o
[build] [31/123  25% :: 38.403] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoHollow.cpp.o
[build] ninja: build stopped: subcommand failed.
[build] Build finished with exit code 1
```

bubnikv, 2021-09-25:
We could only support our Appimages and self compiled using our static dependencies.
We have updated our build instrcutions:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

> 2.3.1+UNKNOWN is just the one version that is provided in pamac in the Manjaro user repository:

You have to check with pamac. We don't have resources to support each and every Linux distro.

Regarding your Locales issue, if it is still an issue for you, please open a new ticket specifically to that issue. Please don't mix multiple issues.

closed 2021-09-25 by bubnikv