prusa3d/PrusaSlicer #6548: Prusa Slicer 2.3.1 not activating non-modal settings window if settings window is minimized

clivius, 2021-05-23:
### Version
PS Version 2.3.1+win64

### Operating system type + version
Windows 10Pro 2004 build 19041.985
I have dual monitors setup but I'm only working on one monitor with PS.

### 3D printer brand / version + firmware version (if known)
i3MK3S+ firmware 3.10

### Behavior
I have my PS GUI set to use non-modal settings window.  I noticed that If the Settings window is minimized to the taskbar, clicking on the Settings icon at the top of the graphics window or any of the gear icons in the right options window next to Print Settings/Fillament/Printer does not pop up the Settings window.   Neither does pressing Ctrl + 2, 3 or 4.  I have to activate first on the taskbar.  Then if it's behind other windows, using any of the methods above will pop it on top and make it active.

It's very frustrating and I feel like this is a bug.  Using any of those methods should pop up the settings window no matter what.

Thanks/Livius

lukasmatena, 2021-11-08:
I believe this is fixed in PrusaSlicer 2.4.0-beta1. Can we close this?

clivius, 2021-11-08:
Awesome. Thank you!

closed 2021-11-08 by clivius