prusa3d/PrusaSlicer #6551: Program freezes at g-code generation

theozoli, 2021-05-23:
### Version
_Version of PrusaSlicer used goes here_

PrusaSlicer
Version:   2.3.1
Build:     PrusaSlicer-2.3.1+UNKNOWN

from the Ubuntu app store

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Ubuntu 20.04.2 LTS

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.4.0-73-generic x86_64
Total RAM size [MB]: 16,652MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 20.2.6
Vendor:       Intel
Renderer:     Mesa Intel(R) HD Graphics 530 (SKL GT2)
GLSL version: 4.60

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Artillery Genius fw: 3.0.5

### Behavior
* _Describe the problem_
GUI freezes at g-code export at some 3D models, while 1 core of the CPU is fully loaded by the software
the g-code file appears at the selected location, but it is not printable (printer stops after about a 10mm line printed)
The status line indicates that the file is exported.

If I use a profile I used for other prints and insert the actual 3D model ends up with the same result.

I tried the repair STL option from the file menu and the result was the same after using the converted .obj file

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
   
export g-code
  
* _Expected Results_

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  
[image]

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[attachments.zip](https://github.com/prusa3d/PrusaSlicer/files/6528043/attachments.zip)

ppesek, 2021-05-27:
I have the same behavior happening, but with an .STL (also Ubuntu 20.04).

corvairjo, 2021-06-30:
I have the same freezes on Ubuntu 20.01 LTS. It happens every time when I generate g-code. The g-code is fine, but the programm freezes indefinately and mut be forced closed. It did not happen with previous versions of the slicer.

closed 2026-09-29 by prusaslicer-bot[bot]