prusa3d/PrusaSlicer #6552: Printing non-manifold surfaces with arbitrary thickness

coreyfro, 2021-05-23:
### Version
All

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
Not all models make sense as volumes or need to be manifold shells to be valid designs.  A non-manifold surface which is a valid geometric construct has all the data necessary to become a valid geometric object to generate CNC paths; Mesh, normals.
[image]

Trying to arbitrarily modify a non-manifold surface into a volume can be a destructive process, a complicated process, and is an unnecessary step. Below is an example of using blender's 'solidify' or thickening feature.
![Uploading 2021-05-21 23_36_33-Blender_ [C__Users_coreyfro_Documents_Lancia Rear Surface.blend].png…]()

A feature such as printing non-manifold surfaces with arbitrary thickness would be a useful process for any industry adjacent to vacuum forming, mold making, working with topological data either generated procedurally or perhaps collected such as data from USGS.
[image]
[image]
[image]
[image]

This is not intended to repair invalid geometry with inverted normals.  Such would still be the responsibility of the designer.
[image]

Where arbitrary thickness would come in is by selection which normals to expand the surface into to create G-Code paths.  This would necessarily require valid geometry from the user.

An alternative interpretation to satisfy existing slicers would depend on the designer constructing arbitrary simplified surface geometry to produce a manifold surface and then the slicer permitting the removal of certain surfaces.  This would provide the slicer the necessary information to recognized desired perimeter while allowing the designer to print just the desired surface at an arbitrary thickness.

This alternative would be akin to vase printing where top and bottom geometric information is discarded and infill would be replaced with support material. In this image the bottom is removed where as it would be desirable to remove the arbitrary surface opposite the original.
[image]

_Is this a new feature request?_
Yes

tkircher, 2021-06-11:
This is definitely a mistake. No 3D models of any kind should ever be non-manifold. It's not enough to just be orientable.

coreyfro, 2021-06-11:
It isn't a mistake. Surface modeling is a feature in Parametric modelers.

The point is, to make a surface like this to have a volume is a DESTRUCTIVE
PROCESS to the model.

We can interact with the surface and visualize it with the slicer.

Turnoff what you know and answer this.

Why can we not simply generate g-code which traces the surface and, by
selecting which normal to expand from, generate an arbitrary thickness.

tkircher, 2021-06-11:
Why can't you simply generate a correct model? If you want a single layer thickness, use 0.4mm instead of 0mm.

Sineos, 2021-06-11:
Isn't this the same discussion https://github.com/prusa3d/PrusaSlicer/issues/3062?

coreyfro, 2021-06-11:
1. That's not how surfaces work.
2. Refer to the "thickened" model above.

coreyfro, 2021-06-11:
No.

tkircher, 2021-06-11:
@coreyfro It's basically the same. Those models use 2D surfaces to fake a 3D object and are depending on tricking the slicer in just the right way to get it to work. As one of the lead Slic3r devs in that thread commented,

"The 3DLabPrint models are invalid models from the point of computer solid geometry modelling. We would have to implement a special mode for handling such broken models."

So, your broken models are unlikely to be supported, ever.

What might be interesting is a better implementation of the thicken algorithm that Blender uses. But looking at what it does to your model above, it's clear that the model itself is a very poorly generated and low resolution triangulation surface. The basic idea of taking each triangle to be the center plane of a triangular prism of uniform thickness and stitching them together is hard to improve upon though.

coreyfro, 2021-06-11:
This definition of "broken" is arbitrary.  I have SAID it isn't a solid
model, it is a surface.  I am a CNC engineer, a "solid" model is not a
requirement for any other part of the CNC industry.

If you don't have the answer, leave the problem for someone who does.

tkircher, 2021-06-11:
"I have SAID it isn't a solid model, it is a surface."

Then it's a big ask for a program that only supports solid models to support it.

"I am a CNC engineer, a "solid" model is not a requirement for any other part of the CNC industry."

If you were a real engineer, you'd know that shells have a specified thickness, and all thermoplastic molds are solid models. So at some point, whether the computer does it for you or not, your 2D surface is being converted to a 3D volume.

coreyfro, 2021-06-11:
Now you are resorting to ad hominem attacks

You cannot make a surface an arbitrary thickness.  This would require
extruding every face an arbitrary value.  The result is included in my
original feature request. It is a broken model.

When generating g-code, you can follow any path which is 2D. If you
intersect a surface with a plane, you get a path.  That path can be used to
generate g-code.  That path can also be used to generate a new path at an
arbitrary offset.

This logic is already used in the slicer for the surface of a solid model.
The only difference is a solid model, being manifold, also has an infill.

If anything, printing a valid surface is easier because all infill is not
requires.  You just make a path out of the intersection and then support it
with support like anything else.

This process is functional in all slicers as is evident with vase printing.

This is simply a vase with a section removed.  I would GLADLY settle for a
process which permits taking a solid model, and selecting faces to remove
and not be printed IF THE RESULT was a surface of arbitrary thickness.

This process, however, seems more complicated.

Sineos, 2021-06-11:
Well, I think both @tkircher and @coreyfro arguments have some value.

In my view:
- A surface or hull has per definition a thickness of 0
- A thickness of 0 needs to be turned into a volume at some point in time, if we want to have a physical object
- Thickening the surface into a volume, e.g. by Blender, is error prone and complex
- The idea of having this "thickening process" done by actually printing it is kind of cool. If I understood correctly, it would just mean: Follow the path of the contour and lay down X perimeters on the bed

coreyfro, 2021-06-11:
THANK YOU! This is EXACTLY what I am suggesting.

tkircher, 2021-06-11:
@sineos It might be possible to simulate the printing process to produce a new valid model. You orient the object to a reference frame, then take intersections of the surface and a stack of parallel planes, then make a Minkowski sum of the intersection path with a circle or sphere.

Sineos, 2021-06-11:
@tkircher how it could be technically achieved in the slicer is definitively above my pay grade.
In my naive understanding it would be as simple as
- Follow the vector of the contour and extrude. 
- When all vectors are followed, "step to the side" by one extrusion width and repeat. 
- Do this until the desired number of perimeters are printed
- Go to the next layer and repeat above steps

tkircher, 2021-06-11:
The slicer is just the wrong place for this tool. It should only be required to accept valid models. I think the real question is that if people want the convenience of using the computer to generate 3D models from 2D models, what's the best process for that? If you can generate a valid model that slices the same way that you would want the slicer to treat the invalid model, then that's basically solved in the best way.

Sineos, 2021-06-11:
Considering that this ticket is a feature request or feature idea, then I guess it is valid to discuss what would be the tool or approach to take. 
I mean the CNC process works just like this: Follow a contour and take away material. What is left is a surface. 3D printing has a close heritage to CNC, so I would think it is not totally off limits

coreyfro, 2021-06-11:
Slicers already do the work.  They produce nesting paths to approximate the
surface of a solid body. It's still following a surface.  The only
difference is these paths have ends instead of being loops.

The surfaces still have normals. The normals,.instead of representing what
is inside or outside would represent a side a or a side b.

There could be three print modes.
1. Arbitrary thickness to side a
2. Arbitrary thickness to side b
3. Equal arbitrary thickness

Sineos, 2021-06-11:
I'd guess a bit related: https://github.com/prusa3d/PrusaSlicer/issues/6574#issuecomment-851345097

bubnikv, 2021-12-21:
Would you please provide models? I would like to test Cura "Surface Mode" settings.

TheRooster13, 2021-12-21:
Yes, please implement this feature. Slice the model into paths and trace the paths. The thickness can be controlled by changing the extrusion width.
Saying a slicer has to only work on 3d models is arbitrarily limiting. Giving creative designers new ways to make physical models that match their imaginations is what 3d printing is all about! This would be one more tool for their toolbox!

TheRooster13, 2021-12-21:
> Would you please provide models? I would like to test Cura "Surface Mode" settings.

Here is a model that I wish I could have printed using surface mode a year ago or so.
[Fantome_Shell.zip](https://github.com/prusa3d/PrusaSlicer/files/7757131/Fantome_Shell.zip)

Here is a quick model I made as an example of a topo surface. It would need to be printed on its side. This is one that I would like to see printed in a back-and-forth pattern instead of traveling to always start on one side.
[Example_Topo.zip](https://github.com/prusa3d/PrusaSlicer/files/7757188/Example_Topo.zip)

ledvinap, 2021-12-22:
I'm afraid this will be quite complicated with Slicer implementation. All code assumes that you get _closed_ polygons with well defined interior after slicing.
Simple offsetting of open polylines will result in overlapping segments, and there is little framework to clean this (maybe thin-wall does a little bit).

IMO only viable approach would be to offset polylines generated by slicing (creating polygons), union them and continue normal processing. Clipper does not support unidirectional offset (?), so this operation will grow surface in both directions. Even with this limitation, it may be useful in some cases ...

dartrax, 2021-12-22:
fyi, from what I understand the new CraftWare Pro slicer can do this. It is called „blade wall“: https://help.craftbot.com/forum/topic/thin-wall-not-created/

coreyfro, 2026-09-18:
This is a trivial feature that should really get kept active.