prusa3d/PrusaSlicer #6555: Prusa Slicer - Creality Ender 3 profile - X and Y are reversed.

rhandel, 2021-05-26:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
WIN 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Creality 3Ender 3

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Slices perfectly but reverses X and Y

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[3DBenchyEnder3_V1.zip](https://github.com/prusa3d/PrusaSlicer/files/6544052/3DBenchyEnder3_V1.zip)

rtyr, 2021-05-26:
Please provide some description of the issue.

rtyr, 2021-05-26:
So the preview is OK, but printer prints it differently? It is a problem on the printer side.

rtyr, 2021-05-26:
Likely incorrectly connected X and Y motor (X connected to Y and vice versa)

rhandel, 2021-05-26:
I checked.  They are connected correctly.
The only thing that looks "reversed" is the waste line printed along the 
Y Axis not the X.  Is there a way to move that?  It nearly clobbered the 
clips that hold down the build surface.

Thanks for the reply,

Robin

rtyr, 2021-05-26:
> The only thing that looks "reversed" is the waste line printed along the Y Axis not the X.

So nothing is reversed after all. Intro line along the Y axis is normal and as far as I know, it is printed along the Y axis with official Creality profiles (Cura) as well.

> It nearly clobbered the clips that hold down the build surface.

Stock Ender 3?

> Is there a way to move that?

Yes, you can edit start g-code section.

[image]

rhandel, 2021-05-26:
Perfect.  Thank you.  Or I can move the clips to the front and back rather than the sides.  I wasn't sure that was the only issue with the Ender 3 profile but it seems that's all that was wrong.

Thank you for the assistance.

RAH

closed 2021-05-26 by rhandel