prusa3d/PrusaSlicer #6557: Feature Request: Drawing straight-line pain-on supports and seams

mix579, 2021-05-26:
### Version
N/A

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
I love paint-on supports and paint-on seams but it's very hard to draw straight lines. In https://github.com/prusa3d/PrusaSlicer/issues/5970, @DaveTSG suggested a way to draw straight vertical lines for paint-on seams. I'd like to request a generalized mechanism that can be applied to both seams and supports. 

For example, pressing the Alt-Key with a left mouse click should "lock" movement in in whatever direction the mouse is initially moved (as projected on the surface the click originated on). 

Alternatively, a way to select to points and then draw a line between. Similar to the option in Fusion 360 for measuring distance between two points.

Caveat: I'm sure there are cases in which such selections will be ambiguous due to projecting a 3D model onto a 2D display but 

_Is this a new feature request?_

Yes

Related topic:
https://github.com/prusa3d/PrusaSlicer/issues/5970

The-Old-Tom, 2021-12-08:
Not sure if it a glitch or by design but in 2.4.0-beta3 you can use the view clipping to draw straight line by painting on, shifting the clip, then removing the highlight.
It would be much nicer if there was something like a modifier block or straightedge brush to make placing the seam more straightforward and accurate though.

dartrax, 2021-12-08:
> Not sure if it a glitch or by design 
It‘s intended behavior (by design).

mix579, 2021-12-08:
It's a workaround that may work under certain circumstances (if the clipping occurs at the appropriate angle) but a long shot from a generally -- and easily -- applicable solutions as requested in the original post.

The-Old-Tom, 2021-12-08:
> It‘s intended behavior (by design).

Good to know

Though I have to agree with mix579 that it isn't a complete solution for the original request.

dartrax, 2021-12-08:
Yes, that's true, and it might also not work because only one border (the clipped one) will be straight, see here:
https://github.com/prusa3d/PrusaSlicer/issues/7059#issuecomment-942144953

A better workaround is here:
https://github.com/prusa3d/PrusaSlicer/issues/7059#issuecomment-934117523

I assume this one will be closed as duplicate of #7059.

lukasmatena, 2024-02-19:
Closing as a duplicate of #5970.

closed 2024-02-19 by lukasmatena