prusa3d/PrusaSlicer #6563: Bottom 3-4 layers of print tapering, causing heavy seem in split up models

Prototypers-lair, 2021-05-27:
### Version
2.3.1

### Operating system type + version
Windows 10 pro build 19042.985

### 3D printer brand / version + firmware version (if known)
Anycubic i3 mega running custom [Marlin 1.1.9](https://www.thingiverse.com/thing:3249319)

### Behavior
I cut a model in half because it is higher than my Z height. In 3D it all looks perfect, but when I print the model, the first 3 layers flare out. It's hard to capture, but I'll try and explain. I cut the model in half and printed each piece separately. In the layer view in PrusaSlicer, the model looks fine (starts as a flat surface, and builds from there). In reality, the model starts with a slightly smaller surface area on the bottom layer, and flares out over 3 layers. Almost as if the slicer is trying to create a wedge to make build-plate separation easier. 

I don't know if this is a slicer bug or a setting I'm not seeing. I have printed flat parts before on this printer with this FW, so I don't expect it to be a FW issue.

I have checked if I accidentally enabled "elephant's foot compensation", I did not. It is not an adhesion issue, because I checked the print, periodically, and it was not lifting. I have attached pictures to illustrate:

[image]
[image]
[image]

#### Project File (.3MF) where problem occurs

[Deadpool top.zip](https://github.com/prusa3d/PrusaSlicer/files/6552492/Deadpool.top.zip)

rtyr, 2021-05-27:
I don't see any issue in the g-code, but your first layer extrusion width seems to be set incorrectly. (it is calculated over layer height, so 110% = 0.33mm)

[image]

Prototypers-lair, 2021-05-28:
> I don't see any issue in the g-code, but your first layer extrusion width seems to be set incorrectly. (it is calculated over layer height, so 110% = 0.33mm)
> 
> [image]

I thought the first extrusion width was as a percentage of default extrusion width (110% of 0,4 = 0,44mm). Are you sure this is % of layer height?

rtyr, 2021-05-28:
> I thought the first extrusion width was as a percentage of default extrusion width.

No

> Are you sure this is % of layer height?

Yes, I am sure. Read the tooltip. It is also clearly visible in the preview.

[image]

Prototypers-lair, 2021-05-28:
I just saw that. I modified it, and the first layer looks a little better, it is still flaring out in layer 1-3 rather than going straight up

rtyr, 2021-05-28:
The g-code output from the PrusaSlicer is OK, you will likely have to tune up your first layer on the printer side. It can be also affected by cooling fan, which turns up 100% on the 2nd layer and may temporarily cause significant temperature drop. I suggest to use "full fan speed at layer" option and set it to 3-4 (see the screenshot). I should also mention your negative XY compensation, but I suppose it was used on the bottom part as well.

I am closing this issue. The g-code output is OK, so is the test print of your g-code.

[image]

[image]

closed 2021-05-28 by rtyr