prusa3d/PrusaSlicer #6564: Mac Build Failed: ld: library not found for -lzstd

ctilley83, 2021-05-27:
### Version
version_2.4.0-alpha0-409-g0b16ffaa0

### Operating system type + version
OSX Big Sur 11.3.1

### Behavior
```
ld: library not found for -lzstd
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [openvdb/cmd/vdb_print] Error 1
make[4]: *** [openvdb/cmd/CMakeFiles/vdb_print.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dep_openvdb-prefix/src/dep_openvdb-stamp/dep_openvdb-build] Error 2
make[1]: *** [CMakeFiles/dep_openvdb.dir/all] Error 2
```
_Is this a new feature request?_
Build Error

### Description:
 compiler is complaining that zstd libraries are not found although zstd 1.5.0 is installed on the system.

ctilley83, 2021-06-10:
Just out of curiosity, can I set a flag to tell the compiler which zstd to use?

brettvitaz, 2021-06-28:
This worked for me:

```
export LDFLAGS="-L$(brew --prefix zstd)/lib"
```

There's probably a much better way to do this, but I'm pretty novice at this toolchain.

destinybonavita, 2021-08-06:
> This worked for me:
> 
> ```
> export LDFLAGS="-L$(brew --prefix zstd)/lib"
> ```
> 
> There's probably a much better way to do this, but I'm pretty novice at this toolchain.

This doesn't seem to be working for me. Would you mind sharing an example script on how you've got it to work? 
I tried adding that to my bash_profile as well with no luck unfortunately.

destinybonavita, 2021-08-07:
I had also tried running `make LDFLAGS="-L/usr/local/opt/zstd/lib"` with no luck unfortunately.

willson556, 2021-08-07:
I was able to get it to build with:
```sh
LDFLAGS="-L$(brew --prefix zstd)/lib" cmake .. -G Ninja
ninja
```

Jnesselr, 2021-08-31:
I ran into this problem as well and the LDFLAGS trick didn't work for me. I ran `make VERBOSE=1` to see what the actual command that failed was, then changed directories temporarily ran the failing command with `-L/usr/local/opt/zstd/lib` at the end, then jumped back and ran make (example for what to look for, running this command directly won't work as it has my paths in it):

```
$ cd dep_OpenVDB-prefix/src/dep_OpenVDB-build/openvdb/cmd/
$ /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -mmacosx-version-min=11.3 -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/vdb_print.dir/openvdb_print.cc.o -o vdb_print  ../libopenvdb.a /Users/jnesselr/dev/PrusaSlicer/deps/build/destdir/usr/local/lib/libboost_iostreams-clang-darwin120-mt-x64-1_75.a -lbz2 -llzma -lz -lzstd /Users/jnesselr/dev/PrusaSlicer/deps/build/destdir/usr/local/lib/libboost_system-clang-darwin120-mt-x64-1_75.a /Users/jnesselr/dev/PrusaSlicer/deps/build/destdir/usr/local/lib/libHalf-2_5.a /Users/jnesselr/dev/PrusaSlicer/deps/build/destdir/usr/local/lib/libblosc.a /Users/jnesselr/dev/PrusaSlicer/deps/build/destdir/usr/local/lib/libtbb_static.a /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/lib/libz.tbd -L/usr/local/opt/zstd/lib
```

ConstantijnCrijnen, 2022-11-28:
Sorry to push this again. I'm getting the exact same error as @ctilley83 after doing a fresh build, following the [docs](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md):
```
[ 96%] Linking CXX executable vdb_print
ld: library not found for -lzstd
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [openvdb/openvdb/cmd/vdb_print] Error 1
make[4]: *** [openvdb/openvdb/cmd/CMakeFiles/vdb_print.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dep_OpenVDB-prefix/src/dep_OpenVDB-stamp/dep_OpenVDB-build] Error 2
make[1]: *** [CMakeFiles/dep_OpenVDB.dir/all] Error 2
make: *** [all] Error 2
```
[Full output here after running make for the first time](https://app.warp.dev/block/O8ngCXAtr2yI2aTXxW9a1A)
[Full output here after running make for any subsequent time](https://app.warp.dev/block/FFBh1mlk0qR4r1Keq9vmlN)

Operating system is macOS Ventura 13.0.1 running on Intel.

The fixes as stated above didn't work for me, however I'm not sure I fully understand @Jnesselr 's solution as I'm still learning about C++/make/cmake

iammattcoleman, 2023-06-14:
I recently made a small change (#10591) and had to work around this issue in order to be able to do builds. I've opened a pull request with a potential fix, but I'm really hoping a CMake expert will review it because I bet there's a better way to solve this.

@ctilley83 @destinybonavita @ConstantijnCrijnen If you're still having issues, you could try my [use-homebrew-zstd](https://github.com/iammattcoleman/PrusaSlicer/tree/use-homebrew-zstd) branch.

closed 2026-09-29 by prusaslicer-bot[bot]