prusa3d/PrusaSlicer #6565: Wrong German transalation of "complete individual objects" checkbox in settings.

mi3dke, 2021-05-29:
### Version
_Version of PrusaSlicer used goes here_2.3.1
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Windows 10 Pro (German)

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Prusa i3 MK3S+

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

Incorrect and misleading translation of the setting "complete individual objects" to "Kompatible Einzelobjekte".
You can not even guess from the naming what the function does.
Instead it should read "Objekte nacheinander drucken" or "Objekte einzeln drucken" to describe the function.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fe60, 2021-06-05:
@mi3dke I proposed "Objekte nacheinander drucken" in the pull request.

This should be adjusted in the documentation too.
https://help.prusa3d.com/de/article/sequentielles-drucken_124589

fe60, 2021-07-28:
Fixed (see comment in pull request https://github.com/prusa3d/PrusaSlicer/pull/6600). 
Screenshot shows version 2.3.3
[image]

closed 2021-07-28 by lukasmatena