prusa3d/PrusaSlicer #6568: Wipe tower printing on air

assumptionsoup, 2021-05-29:
### Version
2.3.0

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa MMU2

### Behavior
Wipe tower has final layer above clear air. I think this may be related to soluble supports. 

[image]

#### Project File (.3MF) where problem occurs
[skelestruder_dragon_adapter.zip](https://github.com/prusa3d/PrusaSlicer/files/6565391/skelestruder_dragon_adapter.zip)

lukasmatena, 2021-05-30:
[FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)). Closing.

closed 2021-05-30 by lukasmatena

assumptionsoup, 2021-05-30:
@lukasmatena Thank you, I missed that.