prusa3d/PrusaSlicer #6571: Gaps in sliced object

KJEJ, 2021-05-30:
### Version
2.3.1

### Operating system type + version
Windows 10 latest version

### 3D printer brand / version + firmware version (if known)
Mk3S latest

### Behavior
Slicer leaves gaps adjacent to the rib on this object (photo included below)

Standard 'Generic PLA' slicer settings plus 'detect thin walls'

I can't seem to attach the .stl file that this is working with

[image]

[image]

I can forward the .stl if you tell me how to do it.   Three other folks with Prusa slicer have the same problem with this file.  I opened the .stl in Fusion and there are no gaps in the location shown.

neophyl, 2021-05-31:
You need to zip up any stl or 3mf files for them to be allowed to be posted.  
3mf  files created using File>Save Project As are the preferred method rather than the stl as the project contains all your settings which is needed to properly try and duplicate your issue.

KJEJ, 2021-05-31:
Here is the file after slicing
[NANO_-_Stern (1).zip](https://github.com/prusa3d/PrusaSlicer/files/6571432/NANO_-_Stern.1.zip)

KJEJ, 2021-05-31:
Sorry, I did not intend to close it...

neophyl, 2021-06-01:
The gaps are caused by the detect thin walls option.  Of course turning that option OFF also shows areas that don't slice as your part has walls thinner than your extrusion width.

The best option would be to actually redesign your part with proper thickness walls that can be printed.  If that option isn't open to you then the easiest way to get it to print is to turn detect thin walls=OFF and then apply a Advanced>Slicing>XY Size Compensation of 0.05mm.  That will thicken the part enough to allow it to print with the currently configured extrusion widths.

You could also reduce your extrusion widths down to 0.4 without using the XY compensation and still with Detect thin walls=Off and that will also slice.

Basically detect thin walls is a crutch that can cause as many issues as it solves, its always best to actually have geometry that can be printed at your chosen nozzle/extrusion width.

KJEJ, 2021-06-01:
Thanks!

Sent from Ken’s iPhone