prusa3d/PrusaSlicer #6574: Feature Request - Multi Perimeter Spiral Vase Mode

z7ths, 2021-05-30:
Why don't you allow multiple perimeters for spiral vase mode?

I can understand not everyone may like how multi perimeter vase mode looks but it would really help for functional parts that you want to print quickly (continuously / without retraction).

When I search for this feature it seems no slicer has it but I don't find any valid reason why.

The functional limitation of the spiral vase mode is just that it is a continuous extrusion with continuous z increase that does not cross itself and that, within less than one extrusion width difference, follows its own historical path at the position of exactly one z layer height less, right? 

Within the above mentioned rules, you can easily do any number of perimeters and even infill. And if you have infill you can also just allow non vase mode top layers as well (same as you already do with the bottom).

But even just the extra perimeters would really be super helpful. I printed the low poly twisted vase in spiral vase mode with just two perimeters and a 0.8mm nozzle and that thing is solid enough that I can put my full (75kg+) weight both on the top and the side without breaking it (using just simple PETG).

Seriously, it would really make spiral vase mode so much more useful. So pretty please?

Thanks!

Sineos, 2021-05-31:
> The functional limitation of the spiral vase mode is just that it is a continuous extrusion with continuous z increase that does not cross itself and that, within less than one extrusion width difference, follows its own historical path at the position of exactly one z layer height less, right?

Maybe I'm just a bit slow: How do you want to achieve a continuous extrusion with a continuous Z increase on multiple perimeters?

z7ths, 2021-05-31:
Behold my mspaint skillz:
[image]
Just pick any point on the path and make that your start, continuous z height increase so that when it reaches it again, it is one layer height higher... i.e. spiral vase mode, right?

I don't show 4+ perimeter but I hope it should be clear that it can be done the same way.

If you cut a thin part out of a model (0.1mm) you can force the slicer to turn around at that point, basically resulting in a 2 perimeter spiral vase mode print. This is what i used for the 2 perimeter low poly twisted vase print, see:
https://www.youtube.com/watch?v=3qo-ck6yKxE
Now that is not ideal as the slicer will actually make 0.1mm gap, but it works well enough to demonstrate the idea.

rtyr, 2021-05-31:
It doesn't look like vase mode to me. It looks more like looping https://github.com/prusa3d/PrusaSlicer/issues/4924, https://github.com/prusa3d/PrusaSlicer/issues/3328.

z7ths, 2021-05-31:
Vase mode is just continuous extrusion and z increase with a non-intersecting path, right?

If I print a cube using vase mode, it's not vase mode because it's a doesn't look like a vase?? 
Are you the vase mode fashion police? ;-)

But seriously, yes, same continuous path design as #3328 but for spiral vase mode to add multiple perimeters. 

The same path (length and all) can be a single circle just by changing the direction of the segments. 
The printer doesn't care what your vase looks, it's works just fine, as long as the are no intersections.

rtyr, 2021-05-31:
> If I print a cube using vase mode, it's not vase mode because it's a doesn't look like a vase??
> Are you the vase mode fashion police? ;-)

Well, I look at your video and it seems that there is no continuous Z increase, It seems like Z increase is done in one step after finishing the whole loop, therefore it doesn't look like vase mode.

Example of the g-code generated in vase mode:
[image]

Sineos, 2021-05-31:
Nifty idea, although I have to say that one key characteristics of the Vase mode is the missing seam and the looping creates quite a prominent one.

commandertaggart, 2022-06-10:
This is an old thread, but I've been thinking about this recently and if the images in my head are correct, you could do actual vase mode, no seam, with multiple perimeters by spiraling the perimeters, alternating inward and outward on each layer.  One row is printed spiraling perimeters inward, then the next outward, as the extruder is on the inner perimeter from the previous layer.  At the end of that layer, it's on the outside, so the following layer is printed spiraling inward again.

This doesn't cover the requested infill, but I think if you want infill, you don't want vase mode.

Am I wrong? Did I explain that well enough?

timur-tabi, 2024-08-17:
Other slicers allow vase mode with multiple perimeters, so PrusaSlicer can do that too.

timur-tabi, 2026-09-02:
This is still a useful feature to have.