prusa3d/PrusaSlicer #6576: Wedge wall has belt (overextrusion)

nulaft, 2021-05-31:
### Version
2.3.1

### 3D printer brand / version + firmware version (if known)
Stock MK3s firmware 3.10.0-4481

### Behavior
I would like to print a flat wall which has a two perimeter width on one end and which linearly increases in width over its height. Unfortunately, a wedge like that develops a belt at a specific height. Upon looking at the gcode preview I noticed that the belt emerges in the height region where the slicer tries to fill the gap between the external perimeters with two (inner) perimeters even though there is not enough space. It looks like it doesn't lower the extrusion width of the inner perimeters like it does with a single line gap fill. Because of this, too much material is extruded in those regions, resulting in the belt.

[image]
[image]
[image]
[image]

* _Steps needed to reproduce the problem_
Slice [this stl file (wedge.stl)](https://github.com/prusa3d/PrusaSlicer/files/6568504/wedge.stl.zip) with a default profile for your favorite material. I used "0.20mm QUALITY" with "Generic PLA" and used some generic PLA. I calibrated the extrusion multiplier for this material using the single wall cube vase mode method (no adjustment to the extrusion multiplier was necessary as the wall thickness was spot on).

* _Expected Results_
Nice flat walls

* _Actual Results_
Flat walls with a ring at a specific height

* _Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[wedge.stl](https://github.com/prusa3d/PrusaSlicer/files/6568504/wedge.stl.zip)
[wedge.3mf](https://github.com/prusa3d/PrusaSlicer/files/6568738/wedge.3mf.zip)

rtyr, 2021-05-31:
It is known problem, see https://github.com/prusa3d/PrusaSlicer/labels/gap%20fill. We plan to improve the behavior.

Enabling "Detect thin walls" option may solve the issue in this case.

nulaft, 2021-05-31:
Thanks a lot for your quick response.

"Detect thin walls" does in fact mostly solve the issue. It replaces the two overlapping internal perimeters with a single gap fill extrusion.
However, there is still a slight bulge where it transitions from gap fill to two perimeters. I am not sure if that transition is too early (internal perimeters are still somewhat overlapping) or if the gap fill needs some kind of extrusion multiplier calibration (is there a parameter for that?).

Anyways, I'm looking forward to future improvements on that topic!

nulaft, 2021-05-31:
I have tested "External perimeters first" and together with "Detect thin walls" it solves my problem perfectly. The walls are flat all around.
I think there needs to be a way to calibrate something such that it doesn't matter if external perimeters are printed first or last.

closed 2026-09-29 by prusaslicer-bot[bot]