prusa3d/PrusaSlicer #6578: No Control over Retracted vs Un-retracted Travel

CCS86, 2021-05-31:
### Version
2.3.1

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
UMO+ / RRF

### Behavior
There seems to be no way to control whether PS uses retracted or un-retracted travel (combing). IMO using un-retracted travel for short moves is a great technique. But, once those travel moves get longer, the inevitable ooze that occurs degrades print quality and causes under-extrusion at the beginning of the next printing move.

There is currently a setting for "minimum travel after retraction", for which I believe the tooltip is wrong:
[image]

I believe that this setting is saying that after a retracted travel move, you need to make at least this value of printing moves before another retraction is allowed. So, if you make a retracted travel, then print a 1mm long move followed by a travel, and your _minimum travel after retraction=_ 2mm, the travel move would be un-retracted.

The incorrect tool tip seems easy enough to prove: set it to 0mm. According to the current tool tip, this would force **all** travel moves to be retracted. Or set it to 10000mm and no retractions should be triggered.

Here is a video showing a very inconsistent choices between retracted and un-retracted travels, with that setting = 0mm. There are small un-retracted travels (great), then longer retracted travel (great), then long un-retracted travel (not great).

https://user-images.githubusercontent.com/12439543/120222239-3da85880-c205-11eb-916d-2c624614f98f.mp4

[Travel.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/6571461/Travel.3mf.txt)

rtyr, 2021-05-31:
Disable the "Only retract when crossing perimeters" option.
[image]

Can we close the issue?

CCS86, 2021-05-31:
That seems to be the root of this. I didn't realize there was a lonely retraction setting in the print settings area. Thanks.

closed 2021-05-31 by CCS86