prusa3d/PrusaSlicer #6580: Artillery Genius doesn't start the print with default profile

Alpacad, 2021-06-01:
### Version
Versione 2.3.1+win64

### Operating system type + version
Windows 10 Pro 21H1

### 3D printer brand / version + firmware version (if known)
Artillery Genius - stock

### N.B.
I do not own this printer, it's a friend of mine's. I will refer everything to him.

### Behavior
* _Describe the problem_: after heating up and priming the nozzle, the head moves to the center to start printing, but then nothing happens.
* _Steps needed to reproduce the problem_
  * Default config for Artillery Genius in Prusaslicer, slice any stl and then start the print
* _Expected Results_: printer starts the job without problem
* _Actual Results_: blob on the bed due to the heated nozzle and gravity

### Notes
This happens with the start gcode of the deafult config file. The problem doesn't occur when using another start gcode I wrote for one of mine printers.

rtyr, 2021-06-01:
Start g-code looks OK, appart from M900/M200 commands which may be not supported by stock firmware.

Can you try the g-code I attached? (it is 4min PLA print).
[Box_0.2mm_PLA_Genius_4m.zip](https://github.com/prusa3d/PrusaSlicer/files/6576828/Box_0.2mm_PLA_Genius_4m.zip)

rtyr, 2021-06-01:
@SzabolcsHornyak Can you check it please? 

(author of Artillery bundle).

SzabolcsHornyak, 2021-06-02:
Im very far from my printer so i couldnt check now. But I asked the FB community and they confirmed. 
Please remove M900/M200 command from here:

[printer:*common_STOCK_FW*]inherits = *common*start_gcode = ; Initial setups\nG90 ; use absolute coordinates\nM83 ; extruder relative mode\n**M900 W[extrusion_width] H[layer_height] D[filament_diameter]\nM200 D0 ; disable volumetric e\n**M220 S100 ; reset speed factor to 100%\nM221 S100 ; reset extrusion rate to 100%\n\n; Set the heating\nM190 S[first_layer_bed_temperature]; wait for bed to heat up\nM104 S[first_layer_temperature]; start nozzle heating but don't wait\n\n; Home\nG1 Z3 F3000 ; move z up little to prevent scratching of surface\nG28 ; home all axes\nG1 X3 Y3 F5000 ; move to corner of the bed to avoid ooze over centre\n\n; Wait for final heating\nM109 S[first_layer_temperature] ; wait for the nozzle to heat up\nM190 S[first_layer_bed_temperature] ; wait for the bed to heat up\n\n; Return to prime position, Prime line routine\nG92 E0 ; Reset Extruder\nG1 Z3 F3000 ; move z up little to prevent scratching of surface\nG1 X10 Y.5 Z0.25 F5000.0 ; Move to start position\nG1 X100 Y.5 Z0.25 F1500.0 E15 ; Draw the first line\nG1 X100 Y.2 Z0.25 F5000.0 ; Move to side a little\nG1 X10 Y.2 Z0.25 F1500.0 E30 ; Draw the second line\nG92 E0 ; Reset Extruder\nM221 S{if layer_height&lt;0.075}100{else}95{endif}

rtyr, 2021-06-02:
Fixed with configuration update 0.0.2.

@Alpacad Please tell your friend to install configuration update (start PrusaSlicer at least twice and then click on the notification in the bottom right corner).

Closing.

closed 2021-06-02 by rtyr