prusa3d/PrusaSlicer #6583: Prusa Slicer (stable branch) freezes after "Slice now"

pmjdebruijn, 2021-06-02:
### Version
`2.3.1+10~g90fd22f85` (stable branch)

### Operating system type + version
xubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
PrusaSlicer freezes after pressing Slice now, though the gcode output is written to disk in it's entirety.

This issue does not occur in 2.3.1 (proper release).

It's my suspicion the issue was introduced by 14efdca4b4134edd18f2f95bd3ff57c47d58fb5f (cherrypick).

bubnikv, 2021-06-02:
I asked @kocikdav to look into it.

BTW it may help if you run the development builds with ASAN enabled
(run cmake with -DSLIC3R_ASAN=on)
You will get crashes where the invalid memory access happens, not much later.

kocikdav, 2021-06-21:
@pmjdebruijn Thank you for reporting.

closed 2021-06-21 by kocikdav