prusa3d/PrusaSlicer #6584: Opening a .3mf file that has a filament profile name same as built-in profile that's not installed selects the wrong filament

Ringarn67, 2021-06-02:
### Version
2.3.1

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Not a printer issue

### Behavior
As Title, there is a thread here, https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/problems-printing-wide-cup-with-tpu/

If I open up a 3mf with a filament profile name which is the same as a system profile PS choose a random filament profile name.
To replicate:
This is NOT easy to explain with my badly written english

Open PS
Add shape, a cube
Install "Smartfill Wood" and use that filament to slice the cube
Change volumetric speed, just to change something
Save new filament profile as "Smartfill Wood - Copy"
Save project as test.3mf
Close PS
Unzip test.3mf
Edit "Slic3r_PE.config", renamed filament_settings_id, line 78, from "Smartfill Wood - Copy" to "Smartfill Wood"
Save everything
Open the file in PS both WITH and WITHOUT "Smartfill Wood" installed and see what happens

Also attached tinker_swiss.zip which is the file where it all begins, try open that file WITH and WITHOUT the built in profile "SemiFlex or NinjaFlex 85-98A" installed

_Is this a new feature request?_
Never seen it before

#### Project File (.3MF) where problem occurs
[TEST.zip](https://github.com/prusa3d/PrusaSlicer/files/6585918/TEST.zip)
[tinker_Swiss.zip](https://github.com/prusa3d/PrusaSlicer/files/6586057/tinker_Swiss.zip)

ssill2, 2021-06-02:
I verified this issue as the reporter describes it.  On my machine it selects E3D Edge as the filament.  Interestingly on a build of 2.4 I built from source, vb_print_regions branch, it temporarily installs the missing profile and flashes up a notification to that effect.  So the issue doesn't happen in 2.4.0.

rtyr, 2021-06-02:
Yes, this problem is already fixed on current master. The fix will be included in PS2.4.0. Closing.

closed 2021-06-02 by rtyr