prusa3d/PrusaSlicer #6585: G-code generation error. Part of object is missing, prints top of print in thin air (customized_sanding_stick)

faern, 2021-06-02:
### Version
2.3.1 (Both on Linux via flatpak and on Windows 10)

### Operating system type + version
Fedora 33 and Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MINI+

### Behavior

* Import the attached STL file named `customized_sanding_stick_15.0_12.0_60.0_wedge_base_45.0_0.0_wedge_base_10.0_3.0.stl` (https://www.thingiverse.com/thing:2404850)
* Slice it. The settings does not matter. The same things happens on all layer heights and filament types etc.
* One of the screws has the head and the part with the threading, but the part in between is missing. The threads are hanging in thin air. See attached screenshot. This is not a rendering issue as printing this generated g-code actually does not try to print the middle part of the screw but instead it prints the threaded part in the air.

[image]

#### Project File (.3MF) where problem occurs
[customized_sanding_stick_15.0_12.0_60.0_wedge_base_45.0_0.0_wedge_base_10.0_3.0.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6586944/customized_sanding_stick_15.0_12.0_60.0_wedge_base_45.0_0.0_wedge_base_10.0_3.0.stl.zip)
[customized_sanding_stick_15.0_12.0_60.0_wedge_base_45.0_0.0_wedge_base_10.0_3.0.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6586957/customized_sanding_stick_15.0_12.0_60.0_wedge_base_45.0_0.0_wedge_base_10.0_3.0.3mf.zip)

ssill2, 2021-06-02:
I can confirm I see that on 2.3.1.  On 2.4.0 it doesn't happen.

ssill2, 2021-06-02:
Here's a screenshot on 2.4.0
[image]

rtyr, 2021-06-02:
Handling of OpenSCAD STLs has been fixed on master with 9c5dd35, the fix will be included in the PS2.4.0.

```
Fixing model import of models with nearly degenerate triangles, which
degenerate after transformation (translation or rotation).

There was certainly a bug in PrusaSlicer, however this bug was triggered
mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
number of decimal digits when saving an ASCII STL, STL export should contain single precision floats

```
Closing.

closed 2021-06-02 by rtyr