prusa3d/PrusaSlicer #6587: Python PostProcessing Script: Permission denied!

foreachthing, 2021-06-03:
### Version
version_2.4.0-alpha0-480-g62ba7de3d

### Operating system type + version
Win10

### Behavior
When trying to run my script [(Slic3rPostProcessor)](https://github.com/foreachthing/Slic3rPostProcessing/blob/master/SPP-Python/Slic3rPostProcessor.py) it gives me the PrusaSlicer Error Code 1.
Then my script outputs this:

```
[Errno 13] Permission denied: 'c:\\users\\USERNAME\\appData\\local\\Temp\\.19932.gcode'
Traceback (most recent call last):
  File "c:\dev\Slic3rPostProcessing\SPP-Python\Slic3rPostProcessor.py", line 156, in main
    with open(sourcefile, "w") as WRITEFILE:
PermissionError: [Errno 13] Permission denied: 'c:\\users\\USERNAME\\appData\\local\\Temp\\.19932.gcode'
```

Only when I export the gcode first, then run the script afterwards it works. It does not on the temp file.

Full script in PS:
`"C:\Program Files\Python39\python.exe" "c:\dev\Slic3rPostProcessing\SPP-Python\Slic3rPostProcessor.py" --xy --noback --p;`

bubnikv, 2021-06-03:
Is your post-processing script opening the file for writing? If so, it may be some process, which is watching over the file and opening it for reading. For example Windows Explorer does that to show a thumbnail.

foreachthing, 2021-06-03:
> Is your post-processing script opening the file for writing?

Well, yeah! How else am I supposed to change it?

> If so, it may be some process, which is watching over the file and opening it for reading. For example Windows Explorer does that to show a thumbnail.

Huh?! Why would it lock it for minutes on end? I even tried it after a while ... something else is going on.
Does it work on your end?

bubnikv, 2021-06-04:
I have a suspicion that @enricoturri1966 's implementation of showing G-code through memory mapping of the G-code may be the culprit here.

foreachthing, 2021-06-04:
Something is going on:
[image]

The temp file is still locked - even minutes after creating.
Only after closing PS the file is freed.

enricoturri1966, 2021-06-07:
I can confirm that when you try to export the gcode, the temporary file is locked by the code used to show the gcode window.
I'll try to find a fix for it.

enricoturri1966, 2021-06-07:
@foreachthing
Could you please check if 8e7591994849352d19f0d94542b1f99c6bea86bc fixes this issue ?

foreachthing, 2021-06-07:
@enricoturri1966 👍🏼 🥇 
Awesome!! Thanks a bunch!

(So, I guess nobody is doing postprocessing but me?!)

closed 2021-06-07 by foreachthing