prusa3d/PrusaSlicer #6588: Uncaught exception: boost::filesystem - PrusaSlicer appimage fails to run.

AndKe, 2021-06-03:
### Version
2.3.1 release

### Operating system type + version
Ubuntu 21.03

### 3D printer brand / version + firmware version (if known)
Custom, Marlin.

### Behavior

```
$ ./PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage 
[2021-06-03 15:42:05.935683] [0x00007f0b3b5935c0] [error]   Uncaught exception: boost::filesystem::copy_file: Invalid argument: "/tmp/.mount_PrusaSBoU17e/usr/bin/bin/../resources/profiles/Anycubic.idx", "/home/andre/.config/PrusaSlicer/cache/Anycubic.idx"
Unhandled unknown exception; terminating the application.

```

#### Project File (.3MF) where problem occurs

bubnikv, 2021-06-03:
Is your "/home/andre/.config/ writable?

AndKe, 2021-06-03:
Yes, I even renamed,  "/home/andre/.config/PrusaSlicer"  and then the application created new directories but failed at the same spot.

bubnikv, 2021-06-03:
Aren't you using some special file system type? Compressed, networked ...

AndKe, 2021-06-03:
no, nothing extraordinary, home directory encryption that is default in Ubuntu.
Also, FWIW:  appimage for 2.3.0 woks fine as always..  only 2.3.1 is affected.

bubnikv, 2021-06-04:
@AndKe The boost::filesystem functions we are using are quite fragile, sensitive to quirks of platform dependence and various file systems. The behavior may change with each boost version.

I tested on WSL2, no problem here. We shall test on Ubuntu. What Ubuntu version?

bubnikv, 2021-06-04:
@AndKe I strongly suspect the home encryption. Why I don't know. For example, we have experienced that some file systems refuse to allow changing user ownership or file attributes (for example the Linux emulation on ChromeOS).

@kocikdav Would you please try to run PrusaSlicer with an encrypted home?

AndKe, 2021-06-04:
@bubnikv , have in mind that if I delete the config directory, it makes a new one, makes files and directories inside, then fail.
missing write permissions make no sense once you see the fresh-created .ini  - please see what it does before failing;:
[PrusaSlicer-this is where it stops.zip](https://github.com/prusa3d/PrusaSlicer/files/6596624/PrusaSlicer-this.is.where.it.stops.zip)

bubnikv, 2021-06-04:
Thanks. I have asked my college to try the encrypted home directory
scenario on his system. Let's see whether he will be able to reproduce the
issue.

nobodyinperson, 2021-06-07:
I can confirm this on Manjaro and I am using ecryptfs for my home directory. Now it makes sense why PrusaSlicer only works with `sudo` for me :facepalm: (`/root` is not encrypted)

bubnikv, 2021-06-08:
Are you guys technical enough to debug it? Maybe running PrusaSlicer with strace to find out which system call fails?

kocikdav, 2021-06-09:
@AndKe  @nobodyinperson  Please would you try this build for us? It will very likely crash same way as before but I hope it will provide more information to us. Thanks.
https://www.dropbox.com/s/wnuxwmwlp9e9p8d/PrusaSlicer-2.4.0-alpha0%2B130-lm-lh-dk-copy-file-linux-x64-asan-gbc1d8a0-202106091911.AppImage?dl=0

nobodyinperson, 2021-06-10:
With this version, upon opening (and accepting the SSL dialog), an error message appears: 

```
Some profiles couldn't be installed because PrusaSlicer is unable to write to the folder "/home/yann/.config/PrusaSlicer-alpha/cache"
```

<details>
<summary>Click to expand the console output</summary>

```
[2021-06-10 13:32:46.449822] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/Anycubic.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/Anycubic.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.455110] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/Artillery.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/Artillery.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.459305] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/BIBO.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/BIBO.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.463771] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/Creality.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/Creality.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.468007] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/FLSun.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/FLSun.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.472050] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/INAT.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/INAT.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.475977] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/LulzBot.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/LulzBot.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.480867] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/PrusaResearch.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/PrusaResearch.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:32:46.484855] [0x00007f775e9bc500] [error]   Copying of file "/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/../resources/profiles/TriLAB.idx" to "/home/yann/.config/PrusaSlicer-alpha/cache/TriLAB.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
=================================================================
==33417==ERROR: AddressSanitizer: heap-use-after-free on address 0x61a000055020 at pc 0x000002120e64 bp 0x7ffc10d55d90 sp 0x7ffc10d55d80
READ of size 8 at 0x61a000055020 thread T0 (slic3r_main)
    #0 0x2120e63 in Slic3r::GUI::SplashScreen::SetText(wxString const&) (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x2120e63)
    #1 0x210b8a6 in Slic3r::GUI::GUI_App::on_init_inner() (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x210b8a6)
    #2 0x210fb1d in Slic3r::GUI::GUI_App::OnInit() (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x210fb1d)
    #3 0x48883a7 in wxEntry(int&, wchar_t**) /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:491
    #4 0x488849a in wxEntry(int&, char**) /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:519
    #5 0x20c6c5f in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x20c6c5f)
    #6 0x4c5ab8 in Slic3r::CLI::run(int, char**) (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x4c5ab8)
    #7 0x480c80 in main (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x480c80)
    #8 0x7f775f6e2b24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)
    #9 0x4b1206  (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x4b1206)

0x61a000055020 is located 928 bytes inside of 1208-byte region [0x61a000054c80,0x61a000055138)
freed by thread T0 (slic3r_main) here:
    #0 0x7f7760cac728 in operator delete(void*, unsigned long) (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/libasan.so.4+0xe1728)
    #1 0x211e8f4 in Slic3r::GUI::SplashScreen::~SplashScreen() (/tmp/.mount_PrusaSh5SZYq/usr/bin/bin/prusa-slicer+0x211e8f4)
    #2 0x4839bcf in wxAppConsoleBase::DeletePendingObjects() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:637
    #3 0x48395ce in wxAppConsoleBase::ProcessIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:443
    #4 0x46337b7 in wxAppBase::ProcessIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appcmn.cpp:397
    #5 0x474f4d2 in wxApp::DoIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/app.cpp:151
[... cut ...]
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==33417==ABORTING
```

</details>

AndKe, 2021-06-10:
Yes, exactly same here 
```
$ chmod +x PrusaSlicer-2.4.0-alpha0+130-lm-lh-dk-copy-file-linux-x64-asan-gbc1d8a0-202106091911.AppImage 
andre@Aranand:~/Downloads$ ./PrusaSlicer-2.4.0-alpha0+130-lm-lh-dk-copy-file-linux-x64-asan-gbc1d8a0-202106091911.AppImage 
[2021-06-10 13:48:04.038255] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/Anycubic.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/Anycubic.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.042221] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/Artillery.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/Artillery.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.044927] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/BIBO.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/BIBO.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.047484] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/Creality.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/Creality.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.050209] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/FLSun.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/FLSun.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.052946] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/INAT.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/INAT.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.055577] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/LulzBot.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/LulzBot.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.058228] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/PrusaResearch.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/PrusaResearch.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
[2021-06-10 13:48:04.061653] [0x00007f75d064c600] [error]   Copying of file "/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/../resources/profiles/TriLAB.idx" to "/home/andre/.config/PrusaSlicer-alpha/cache/TriLAB.idx" failed.
Error message : Invalid argument
Copying was triggered by function: checking install indices
=================================================================
==556002==ERROR: AddressSanitizer: heap-use-after-free on address 0x61a000065e20 at pc 0x000002120e64 bp 0x7ffc4147c770 sp 0x7ffc4147c760
READ of size 8 at 0x61a000065e20 thread T0 (slic3r_main)
    #0 0x2120e63 in Slic3r::GUI::SplashScreen::SetText(wxString const&) (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x2120e63)
    #1 0x210b8a6 in Slic3r::GUI::GUI_App::on_init_inner() (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x210b8a6)
    #2 0x210fb1d in Slic3r::GUI::GUI_App::OnInit() (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x210fb1d)
    #3 0x48883a7 in wxEntry(int&, wchar_t**) /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:491
    #4 0x488849a in wxEntry(int&, char**) /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:519
    #5 0x20c6c5f in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x20c6c5f)
    #6 0x4c5ab8 in Slic3r::CLI::run(int, char**) (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x4c5ab8)
    #7 0x480c80 in main (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x480c80)
    #8 0x7f75d1309564 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x28564)
    #9 0x4b1206  (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x4b1206)

0x61a000065e20 is located 928 bytes inside of 1208-byte region [0x61a000065a80,0x61a000065f38)
freed by thread T0 (slic3r_main) here:
    #0 0x7f75d2900728 in operator delete(void*, unsigned long) (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/libasan.so.4+0xe1728)
    #1 0x211e8f4 in Slic3r::GUI::SplashScreen::~SplashScreen() (/tmp/.mount_PrusaSmN8LaV/usr/bin/bin/prusa-slicer+0x211e8f4)
    #2 0x4839bcf in wxAppConsoleBase::DeletePendingObjects() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:637
    #3 0x48395ce in wxAppConsoleBase::ProcessIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:443
    #4 0x46337b7 in wxAppBase::ProcessIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appcmn.cpp:397
    #5 0x474f4d2 in wxApp::DoIdle() /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/app.cpp:151
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==556002==ABORTING
andre@Aranand:~/Downloads$ 

```

kocikdav, 2021-06-10:
Thank you both! Would you please try running PrusaSlicer (2.3.1 release) with --datadir option pointing outside your Home directory? That should allow you at least start the program. Then please try to export any gcode to your Home folder. There is a different copy function inside PrusaSlicer for exporting gcode on Linux so we are curious how this function will go.

nobodyinperson, 2021-06-10:
With PrusaSlicer 2.3.1 (from the official Manjaro repositories), setting `--datadir` to something outside the `ecryptfs` (I used `/tmp/prusa`), opening and everything works!

Exporting gcode to the encrypted home directory leads to an error dialog saying:

```
Copying of the temporary G-code to the output G-code failed. Maybe the SD card is write locked?
Error message: Invalid argument
```

But nothing useful on the console (even with `--loglevel 5`).

kocikdav, 2021-06-11:
Thanks, can I ask you to test few more builds? Please try the same (datadir to tmp and export to the encrypted home) with this build and with --loglevel=5. There should be trace logs to tell us where exactly the copy function fails.
https://www.dropbox.com/s/70te9irmdqmnull/PrusaSlicer-2.3.1%2B18-dk-home-linux-x64-asan-g5014fec-202106111000.AppImage?dl=0

nobodyinperson, 2021-06-11:
```bash
LC_ALL=C ~/Downloads/PrusaSlicer-2.3.1+18-dk-home-linux-x64-asan-g5014fec-202106111000.AppImage --datadir /tmp/prusa --loglevel=5 2>&1
```

It throws the same error dialog when exporting.

Terminal output as file as it's too large for a GitHub comment:

[prusa-2.3.1+18.log](https://github.com/prusa3d/PrusaSlicer/files/6637816/prusa-2.3.1%2B18.log)

AndKe, 2021-06-11:
It might be useful to look for changes since the 2.3.0 release.
Of hundreds of applications I've never seen something hiccup due to ecryptfs, and of only the 2.3.1 fails, it must be some strange change to it, or it's build tools  - ecryptfs is usually transparent to all apps.

bubnikv, 2021-06-11:
> It might be useful to look for changes since the 2.3.0 release.

I suspect we bumped up boost library version. Boost is a multi-platform
library which is supposed to handle multi-platform issues well. But for
example when running PrusaSlicer on Chrome OS / Linux virtual machine, the
virtualized Chrome OS filesystems mounted into the Linux virtual machine do
not like boost::filesystem::copy_file() to set file attributes even though
setting these attributes seems pretty harmless from security point of view.
We don't know what is going on on your encrypted file system, we have to
either reproduce ourselves or provide you with enough debugging facility to
debug it on your system.

If it was all so simple as it seems.

bubnikv, 2021-06-14:
The same issue has already been reported in #4716 and there is a reference to a bug in the boost::filesystem library which the following very fresh pull request is supposed to fix.

https://github.com/boostorg/filesystem/commit/4b9052f1e0b2acf625e8247582f44acdcc78a4ce

kocikdav, 2021-06-14:
@AndKe  @nobodyinperson Please try exporting with this build (with datadir in temp), we hope it will work this time. If it does, we will use this copy function everywhere.
https://www.dropbox.com/s/8p4c5oczevdr4w8/PrusaSlicer-2.3.1%2B20-dk-home-linux-x64-asan-g1cb5b94-202106141241.AppImage?dl=0

AndKe, 2021-06-14:
@kocikdav   
yes: ` ./PrusaSlicer-2.3.1+20-dk-home-linux-x64-asan-g1cb5b94-202106141241.AppImage --datadir /tmp/prusa `  does run, but it's kind of pointless as it does not see/use my configuration (starts as it runs for the first time) 
- am I missing something?

kocikdav, 2021-06-14:
@AndKe Does Exporting to encrypted directory works? We use several copy_file functions inside PrusaSlicer and the one that handles exporting Gcode should work even when target is encrypted linux filesystem. If it works, we will switch all copy file function calls to this one.

nobodyinperson, 2021-06-15:
This new version works perfectly fine. No crash upon exporting to ecryptfs partition. Thanks!

[PrusaSlicer-2.3.1+20-dk.log](https://github.com/prusa3d/PrusaSlicer/files/6654560/PrusaSlicer-2.3.1%2B20-dk.log)

nobodyinperson, 2021-06-15:
>     * am I missing something?

@AndKe This is expected as you give PrusaSlicer a completely new (and empty) directory where no configuration is stored. (`--datadir`).

AndKe, 2021-06-16:
Is the new appimage ready for download somewhere? and: is/will 2.3.1 release be fixed before the next release?

kocikdav, 2021-06-17:
The fix is merged into our stable branch. Here is a link to download a build:
https://www.dropbox.com/s/eot7lqphsbx43as/PrusaSlicer-2.3.1%2B27-linux-x64-asan-g02f97e1-202106170939.AppImage?dl=0

It will be part of next release, until then this build should work for you. It is basically a 2.3.1 with several bux fixes like this one. im closing this issue, please reopen if theres a problem with this build.

closed 2021-06-17 by kocikdav

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta by back-porting the boost::filesystem fix into PrusaSlicer own copy_file implementation.