prusa3d/PrusaSlicer #6591: Error in determining the overhang angle for supports

Alexx40in, 2021-06-04:
### Version
_2.3.1+win64_

_PrusaSlicer-2.3.1+win64-202104161404_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 (build 19041), 64-bit edition_

### Behavior
* _Overhang threshold is not calculated correctly. For an angle on the part of 40 degrees, supports are turned on at a value of 62, for an angle of 50 - at the Overhang threshold, for an angle of 60 - at 36. There is a feeling that in the program the right angle is not 90, but 100 degrees_

Alexx40in, 2021-07-16:
The error is still present. In order to enable supports for an angle of 45 degrees, I must specify an overhang angle of 57 degrees in the support settings. Thus, it is completely unclear how the program should indicate the overhang angle. because 90-45 = 45, but not 57

imnotme, 2021-07-18:
Same Issue here :+1:

bubnikv, 2021-10-05:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3094
Closing this one.

closed 2021-10-05 by bubnikv