prusa3d/PrusaSlicer #6594: [Feature request] Add pyramid generation in cut option for assembly

Syl20-94, 2021-06-04:
When cutting a object, It will be great if we should add pin (such as trunked pyramid, or cone) to being able to align parts on mounting.
this should be an option to the Cut area with parameter for diameter (assuming that angle of the pyramid is frozen).
by using this option it will do two boolean operation : 
- on upper part it will substract pyramid (or cone)
- on lower part  it will add pyramid (or cone) 

by this we should assembly part with a nice aligment (this will be usefull for both FDM and SLA print) 

please tell me if this should be relevant for a next release

rtyr, 2021-06-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1857, https://github.com/prusa3d/PrusaSlicer/issues/5553, https://github.com/prusa3d/PrusaSlicer/issues/2357. It's on our todo list. Closing.

closed 2021-06-04 by rtyr

Syl20-94, 2021-06-04:
I did not find it before opening sorry.
Thank for your reply , I'm looking forward the availability :)