prusa3d/PrusaSlicer #6595: Pause Print -> Resume print Octoprint

morimorida, 2021-06-04:
### Version
2.3.1

### Operating system type + version
Windows 10 64bit 20h2

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
Had MMU  load fail, and forgot to tighten the extruder tension screw, print resumed without loading a filament and started to purge on purge block.

1) I hit pause print on Printer side.
2) try to tighten the tension screw and x-carriage moved (stepper motor was disabled?)
3) after feeding the filament, tightening the tension screws, hit "resume print" on Octoprint side.
4) the print restarted but due to offset of x, the whole print shifted.

What I noticed after I pause print on printer side, I can maneuver to "Settings" and move xyz axis.  I guess it's nice this can be possible but I'm not too sure, when resumed, it'll go back to the xyz coordinate that it was paused at.

In addition, when I pause print on printer side, I would assume that xyz stepper motor to be locked and only be able to move using the menu (settings-> xyz axis).

rtyr, 2021-06-04:
This is not PrusaSlicer related issue. Please open your issue here https://github.com/prusa3d/Prusa-Firmware. Closing.

closed 2021-06-04 by rtyr

morimorida, 2021-06-05:
I just want to confirm.

Since I'm printing from Octoprint initially, the abnormal behavior of pause command AT PRINTER SIDE is not related to PrusaSlicer, correct?

rtyr, 2021-06-05:
Correct.