prusa3d/PrusaSlicer #6597: Min bottom shell thickness calculation does not take into account the first layer thickness

clivius, 2021-06-04:
### Version
PS Version 2.3.1+win64

### Operating system type + version
Windows 10Pro 2004 build 19041.985

### 3D printer brand / version + firmware version (if known)
i3MK3S+ firmware 3.10

### Behavior
In the informational area at the bottom of **Horizontal Shells** section under _Print Settings_, the min bottom shell thickness value seems to only takes into account the layer height.  The same goes for the value of the _Bottom_ field for _Minimum shell thickness_.  It would be helpful to take into account the First layer height value in the calculation for minimum and total shell thickness.

For example for a layer height of 0.3mm and first layer height of 0.2mm, the minimum bottom shell thickness should be 0.5mm not 0.6mm.  Additionally, the bottom shell thickness using four solid layers should be 1.1mm not 1.2 (4*0.3).

I understand that the minimum bottom shell thickness can be user defined but the default min value might want to be (first_layer_thickness) + (layer_thickness).  The informational data below the fields should always be this minimum calculation for bottom shell thickness.

I'm new to the slicer and 3D printing in general and don't know why it is the way it is.  Just pointing out an observation.  I use these values to tweak designs to these exact numbers for clean slicing and it's nice to have them handy without going thru a calculation for the exact values.

Thanks/Livius

rtyr, 2021-06-05:
Shell thickness and minimal shell thickness are two different values, which doesn't have to be equal. The purpose of minimal thickness setting is to prevent issues when using variable layer height. Example: Model printed with base 0.3mm layer height (4 top solid layers are ok) with some sections printed with 0.07mm layer height (4 top layers are not ok). Closing.

clivius, 2021-06-05:
Thanks for the understanding.  However, what you described is not at issue.  A new project defaults to standard defined layers.   Print settings can be set before any object is loaded into the print volume.  Variable layers aren't a factor for a new project and empty print volume.  Even for variable layer heights the first layer is always 0.2mm, the specified value. 

Looking at the attached image and from what you described, only the minimum shell thickness values are enforced when switching to variable layer heights (orange arrows).  For non variable layer heights it appears like they might be just limits because there's tension between the min values and the actual specified layer and first layer heights.  Which calculation wins?  Also, the blue arrows seem to be the defaults and simply informational for the user and they don't change when variable layer heights are used.  

What I'm suggesting is that the calculated Bottom shell value (blue arrows) is false.  The true value is with the inclusion of the first layer height (green arrows).

[image]

rtyr, 2021-06-05:
I see. It is not so simple though, first layer printed on top of the soluble supports (0 z contact distance) is also bottom shell (printed by regular layer height).

clivius, 2021-06-05:
Hmm, good point.  However, I think a high percentage of users don't print on soluble supports.  If you squint at the percentage that do, you might miss them.  How about a message like this for the general population?

"Bottom shell is 1.1mm thick for layer height 0.3mm and first layer height 0.2mm, 1.2mm for layer height 0.3mm. Minimum bottom shell thickness is 0.6mm."

Look at what happens when the layer height is **0.15mm**, fist layer **0.2mm** and **5** layers for Bottom shell.  Currently the Bottom shell thickness is displayed as 0.75mm (0.15*5).  If you're not printing on top of soluble supports the value should be 0.8mm (0.15*4+0.2).   This is also revealed in the layer slider:

[image]

For thicker layers (using larger nozzles) this is even more problematic.  For layer height **0.4mm**, fist layer **0.2mm** and **3** layers for Bottom shell (system defaults), the bottom shell thickness is calculated and displayed to be 1.2mm.  If I slice a 1.2mm thick shell, the slicer will only print a 1mm bottom shell (0.4*2+0.2).  There's a significant difference between 1 and 1.2mm shell thickness.

closed 2026-09-29 by prusaslicer-bot[bot]