prusa3d/PrusaSlicer #6599: Version 2.3.1 clicking "Place on Face" tool takes minutes to compute.

Boorkus, 2021-06-05:
### Version
2.3.1+win64

### Operating system type + version
Windows 10 64 bit Home version 19042.985

i7-9700kf @ 5GHz
32GB DDR4 RAM
GTX 980 Ti 
Samsung Pro 1TB NVME SSD

### Behavior
Drag in .3mf or .stp file to build plate
Click on body
Click on "Place on Face" tool
Wait 1-5 minutes while it pegs my CPU, before the planes eventually become visible to click on to align to build plate. 

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
.3mf file uploaded.

rtyr, 2021-06-07:
Please provide your 3MF project file (it is not uploaded in your initial post).

Boorkus, 2021-06-08:
> Please provide your 3MF project file (it is not uploaded in your initial post).

Apologies, somehow the upload didn't work the first time. Updated post with link.

Boorkus, 2021-06-08:
I think I just found the issue, somehow Solidworks was exporting with a rediculously high precision in the .3MF/.STL export... The 3MF file for some reason is 3mb. I just re-exported the file again and now it's only 30kb, and the performance issue is gone. I don't think this is a bug with PrusaSlicer, I think this is a PEBKAK error.

rtyr, 2021-06-08:
Can we close this issue then?

closed 2021-06-08 by Boorkus