prusa3d/PrusaSlicer #6601: Modificator infill at wrong layer

merc-slk, 2021-06-05:
### Version
_Version of PrusaSlicer used goes here_
2.3.0 win64

### Operating system type + version
Win 10 Pro 20H2 19042.985

### 3D printer brand / version + firmware version (if known)

MK3S and MMU2S updated to latest stable version
 
### Behavior
Infill at 10% Honeycomb or any other, doesn't matter.
Put modificator at 1.40mm (layer height 0.2mm) to change to star with 25% infill
Result new infill happens at 3.00 mm

_Is this a new feature request?_ no

#### Project File (.3MF) where problem occurs

[ISR.zip](https://github.com/prusa3d/PrusaSlicer/files/6602698/ISR.zip)

neophyl, 2021-06-05:
Its working as I would expect.  Although how it works can be a bit confusing.
The modifier is relative to your model.  So placing it 1.4mm up is placing the centre of your modifier 1.4mm above the centre of the model.  For example if you use the drop to bed function the Z position is shown as -1.4mm.  If you want it a specific height above the bed then you need to put in from that offset.  
So for example if you wanted it 1mm above the bed you would put -.4 in the z position.

merc-slk, 2021-06-05:
Thanks,
Is indeed a bit confusing. Should be relative to the build plate form 0 up. I've put Z1.6 and starts at 1.4 with a total layer height of 7.
7/2=3.5 - 1.6 =1.9 -0.2 Layer height 1.7ish  almost.... if it's calculated this way.

neophyl, 2021-06-05:
You could always use a height range modifier instead of a modifier mesh. Those do work over specific height.

davidkerny, 2021-06-07:
Hi! Yes, this function doesn't seem to work right. I've clicked add part > box. Moved the box 20mm up in Z-axis, it raised like 100mm instead of 20 above the center of the model.

[image]

neophyl, 2021-06-07:
That's because of an existing issue to do with scaling.  Once you have scaled or rotated your original part then modifiers act in ways that are not easy to determine.  If you search you will find several relating to modifiers and scaling/rotation.

merc-slk, 2021-06-08:
Disagree. Didn't change size or rotate whatever.  1 on 1 stl file. And even if you rotate it doesn't change the z height.

closed 2026-09-29 by prusaslicer-bot[bot]