prusa3d/PrusaSlicer #6608: Slicer creates arbirtrary top layer which is not in the STL

gfornax, 2021-06-08:
### Version
Version 2.3.1+linux-x64

### Operating system type + version
debian linux testing 64bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
I have the following STL created with openscad:

[image]

when loaded into prusaslicer, it looks like this:
[image]

Now, for 6ish other variations of that part or even this one scaled to 50% everything works like it should.
But for some reason, prusaslicer decides to add a solid top layer out of nowhere here:
[image]

This is obviously a bug. I presume it is no issue with the STL as prusaslicer does not even create a support for that incorrect top layer if forced to do so:
[image]

#### Project File (.3MF) where problem occur
[large_box.zip](https://github.com/prusa3d/PrusaSlicer/files/6619482/large_box.zip)

Thank you in advance!

neophyl, 2021-06-08:
Openscad  is notorious for producing these sorts of issues with models for slicer unless you are really careful. 
You can either modify your scad model to fix the problem, fix it using some other software, like netfabb or wait until PS2.4 is released as that is supposed to be more tolerant of common openscad issues. 

As for the support not being created, do you happen to have don’t support bridges enabled (it is by default) as that would treat that layer as bridging so would generate support.  When I’m back at a pc in about 12 hours I’ll download your project and look at the stl but having seen so many of these now I’m almost certain there are going to be issues with your stl.

gfornax, 2021-06-08:
Interesting. What is weird is that it seems absolutely random if that happens or not. If i only change, e.g., where the walls are located, or scale it a bit, the top layer (which is several layers thick) disappears and everything is fine.
FWIW, your guess with the bridge support was correct, didn't check that. I'll remove that part to avoid confusion as this is not the actual issue here.

rtyr, 2021-06-08:
It's sliced correctly on current master.

Fixed with https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17. Closing.

closed 2021-06-08 by rtyr