prusa3d/PrusaSlicer #6609: Request Cool-down gradient option to reduce warp for Nylon and similar warp prone materials

Sipowers, 2021-06-09:
### Version
2.3.0

Version 2.3.0+win64_

N/A

### Operating system type + version
Windows 10 Home

### 3D printer brand / version + firmware version (if known)
PRUSA i3MK3s - the latest one.

### Behavior
This is not a problem its a feature request.
_Is this a new feature request?_ YES

Regarding Nylon645 and all Nylons in particular.

I have been printing a lot of long thin parts in Nylon recently for a project on my PRUSA using the textured bed and I have done all my research on the settings etc. so please don't bother telling me all the various techniques - because I am aware of them all.

From my experience, most warping occurs during the (un-controlled) cool down of these parts at the end of a print, which due to the nature of Nylon (and some other materials) which tend to contract when cooling rapidly.

My request is this. To add an option to Prusa Slicer that will provide controlled cooling of the heatbed in the Gcode at the end of a print. A variable that controls the cooling in a linear fashion would be fine ie. 2 degrees C per minute etc.

This obviously adds to the total print time but when you are trying to produce precision parts - this really does not matter. But minimising warp - does.

Thanks.

rtyr, 2021-06-10:
That should be easy thing to do using filament specific end-gcode (Filament settings -> Custom g-code (end g-code)). 

Some example:
```
M104 S0 ; turn off hot-end temperature
M190 S{bed_temperature[0]-5} ; cooldown
G4 S90 ; wait 90s
M190 S{bed_temperature[0]-10} ; cooldown
G4 S90 ; wait 90s
M190 S{bed_temperature[0]-15} ; cooldown
G4 S90 ; wait 90s
M190 S{bed_temperature[0]-20} ; cooldown
G4 S90 ; wait 90s
M190 S{bed_temperature[0]-30} ; cooldown
```

As you can imagine, there are many possible variants (each user can have different demands), so I would say this is quite good way how to do it, without adding some separate feature.

rtyr, 2021-06-18:
Answered, no response. Closing.

closed 2021-06-18 by rtyr